ARMmbed/mbed-os-example-blinky #29: compiling problem of exported project from online IDE

MaggieMei, 2016-08-18:
I export the program mbed-os-example-blinky from online IDE to uvision5 on FRDM-K64F. When compiling. it shows the errors as below:

*** Using Compiler 'V5.06 update 1 (build 61)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'mbed-os-example-blinky'
compiling main.cpp...
main.cpp(1): error:  #5: cannot open source input file "mbed.h": No such file or directory
main.cpp: 0 warnings, 1 error
".\.build\uvision5\mbed-os-example-blinky.axf" - 1 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:00

It seems all related mbed-os files are not exported successfully.
I add the mbed-os.lib into the project manually, but still the same error.

MarceloSalazar, 2016-08-18:
Could you try MDK 5.20 with ARM Compiler 5.06 Update 3?

SYuPing, 2016-08-23:
Hi,
I have same problem with @MaggieMei ...

Here is my toolchain version:
MDK-ARM Standard Cortex-M uVision V5.13.0.0

Thanks

MarceloSalazar, 2016-08-23:
Please update to MDK 5.20 (ARM Compiler 5.06 Update 3)

karmasLin, 2016-09-01:
In on-line compiler:
You have to import 2 libraries in online-compiler: "mbed" & "mbed-os", then export to uVision.
Furthermore, in main.cpp you must add following line before compiling in Keil MDK
`#include "rtos.h"`

karmasLin, 2016-09-02:
Refer to [All exporters broken with mbed OS 5 #2591](https://github.com/ARMmbed/mbed-os/issues/2591)
Same problem...

closed 2017-02-02 by 0xc0170