ARMmbed/mbed-os-example-blinky #30: GCC_ARM won't link on NRF51822: cannot move location counter backwards

usedbytes, 2016-08-30:
Won't link when building for NRF51822 like so: (trying to build for micro:bit)

```
$ mbed compile -t GCC_ARM -m NRF51822
Building project mbed-os-example-blinky (NRF51822, GCC_ARM)
Scan: .
Scan: FEATURE_UVISOR
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_IPV4
Scan: FEATURE_IPV6
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
Link: mbed-os-example-blinky
./mbed-os/hal/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM/TARGET_MCU_NRF51_16K_S130/NRF51822.ld:133 cannot move location counter backwards (from 0000000020003d60 to 0000000020003800)
collect2: error: ld returned 1 exit status
[ERROR] ./mbed-os/hal/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM/TARGET_MCU_NRF51_16K_S130/NRF51822.ld:133 cannot move location counter backwards (from 0000000020003d60 to 0000000020003800)
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/user/Programming/mbed/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m NRF51822 --source . --build ./.build/NRF51822/GCC_ARM" in "/home/user/Programming/mbed/mbed-os-example-blinky"
---
```

The same toolchain builds for lpc1768 and lpc11u24 fine (the only other two platforms I have).
Have also reproduced with a different (gcc 5.2.0, newlib 2.2.0) toolchain on ARMv7 built via crosstool-ng.

* OS:  Arch
* arm-none-eabi-gcc package, version 6.1.1-4
 * PKGBUILD: https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/arm-none-eabi-gcc&id=9930233c4e5a722ee63125b2605b928f76b85946
* arm-none-eabi-newlib package, version 2.4.0-3.
  * PKGBUILD: https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/arm-none-eabi-newlib&id=4d6307f2ae9c90f42d32c1a3da82e09f4cade86f

pan-, 2016-08-30:
The target NRF51822 is not supported by mbed-os 5 at the moment. 
It doesn't have enough memory to support the RTOS with the c library used in mbed OS 5.

We are working on reducing memory consumption to make mbed OS 5 available to target with more memory constraints.

usedbytes, 2016-08-31:
OK thanks. Is there any plan to either not list platforms which are not supported, or provide an indication to the user for unsupported platforms?

pan-, 2016-08-31:
It is indicated in `targets.json`: https://github.com/ARMmbed/mbed-os/blob/master/hal/targets.json

The property `release_versions` contain an array of the versions of mbed os supported by each target.

0xc0170, 2017-02-02:
Tested, works. I'll close this issue

closed 2017-02-02 by 0xc0170

danielwegener, 2017-03-05:
Having the same issue here. Took mbed-os-example-blinky, made it build for my JLink/ BLE400/[Core51822](http://www.waveshare.com/wiki/Core51822).

Then I added a single line (`Serial pc...`) which in turn would include the Serial driver:
```c
#include "mbed.h"

DigitalOut led(LED1);
Serial pc(USBTX, USBRX); // tx, rx

// main() runs in its own thread in the OS
// (note the calls to wait below for delays)
int main() {
    while (true) {
        led1 = !led1;
        wait(0.1);
    }
}
```
And I end up with
```
  mbed-os-example-blinky git:(master) ✗ bin/mbed compile -t GCC_ARM -m NRF51822
Building project src (NRF51822, GCC_ARM)
Scan: .
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_BLE
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
Link: src
./mbed-os/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/device/TOOLCHAIN_GCC_ARM/TARGET_MCU_NRF51_16K_S130/NRF51822.ld:133 cannot move location counter backwards (from 0000000020003930 to 0000000020003800)
collect2: error: ld returned 1 exit status
[ERROR] ./mbed-os/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/device/TOOLCHAIN_GCC_ARM/TARGET_MCU_NRF51_16K_S130/NRF51822.ld:133 cannot move location counter backwards (from 0000000020003930 to 0000000020003800)
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /src/mbed-os/tools/make.py -t GCC_ARM -m NRF51822 --source . --build ./BUILD/NRF51822/GCC_ARM" in "/src"

```
@0xc0170 Could you please clarify what part of the issue is "closed" or "works"?

The mbed page pretty clearly announces support for the NRF51822 (see https://developer.mbed.org/platforms/?tvend=11) and it feels a bit disappointing to find out that I could use mbed for this MCU merely to switch LEDs on and off. Could this maybe a problem with the linker (I am not sure where the aforementioned feature flags are coming from)?

- OS: Ubuntu 16.04 (as docker base image)
- gcc-arm-none-eabi-6-2017-q1-update
- mbed-cli 1.0.0