ARMmbed/mbed-os-example-blinky #46: [OOB mbed-os-5.4] Post binary hook not run on make for target MTS_MDOT_F411RE

mray19027, 2017-02-28:
`mbed compile -t GCC_ARM -m MTS_MDOT_F411RE`

Correctly runs post binary hook:
```
            "function": "MTSCode.combine_bins_mts_dot",
            "toolchains": ["GCC_ARM", "ARM_STD", "ARM_MICRO"]
```

`mbed export -i make_gcc_arm`
`make`

Does not run post binary hook and therefore will not run the application

0xc0170, 2017-03-01:
cc @mfiore02 @theotherjimmy

theotherjimmy, 2017-03-01:
This is a know limitation of all of the exporters. None of them can run post-binary hooks.

mfiore02, 2017-03-01:
@mray19027 Yeah, it's a mess. We (MultiTech) don't recommend exporting for the mDot or Dragonfly because the bootloader complicates the issue. Instead, we suggest configuring your IDE/makefile to invoke mbed-cli to compile the application. Then you don't need to worry about any of this.

Last time I took a look at the mbed exporter, there wasn't a way to export pre-build or post-build processes & extra files required by those processes. ( @theotherjimmy confirmed this as I was typing ) That's the main reason things are the way they are now. It would also be a lot for our small team to maintain - we simply don't have the bandwidth.

You can see how the mbed-cli tools handle the post-build process for the mDot below. The snippet is from mbed-os/tools/targets.py in mbed-os-5.3.6.

``` python
class MTSCode(object):
    """Generic MTS code"""
    @staticmethod
    def _combine_bins_helper(target_name, binf):
        """combine bins with the bootloader for a particular target"""
        loader = os.path.join(TOOLS_BOOTLOADERS, target_name, "bootloader.bin")
        target = binf + ".tmp"
        if not os.path.exists(loader):
            print "Can't find bootloader binary: " + loader
            return
        outbin = open(target, 'w+b')
        part = open(loader, 'rb')
        data = part.read()
        outbin.write(data)
        outbin.write('\xFF' * (64*1024 - len(data)))
        part.close()
        part = open(binf, 'rb')
        data = part.read()
        outbin.write(data)
        part.close()
        outbin.seek(0, 0)
        data = outbin.read()
        outbin.seek(0, 1)
        crc = struct.pack('<I', binascii.crc32(data) & 0xFFFFFFFF)
        outbin.write(crc)
        outbin.close()
        os.remove(binf)
        os.rename(target, binf)

    @staticmethod
    def combine_bins_mts_dot(t_self, resources, elf, binf):
        """A hook for the MTS MDOT"""
        MTSCode._combine_bins_helper("MTS_MDOT_F411RE", binf)

    @staticmethod
    def combine_bins_mts_dragonfly(t_self, resources, elf, binf):
        """A hoof for the MTS Dragonfly"""
        MTSCode._combine_bins_helper("MTS_DRAGONFLY_F411RE", binf)
```

We basically do the following:
- write the bootloader binary to the "final binary"
- write 0xFF padding up to 64kB
- write the application binary (starts at a 64kB offset as specified in the linker file)
- compute a crc32 checksum of the application binary
- write the checksum

The checksum is for our bootloader to verify the new application integrity before it is flashed.

0xc0170, 2017-03-02:
@mfiore02 Is this documented as known-issue (present on the platform page?). Exporting should be disabled for this platform then, as not supported with a given reason.

Thanks for the detailed information.

cc @MarceloSalazar (to be aware of this limitations and we shall state that in our OOB)

mfiore02, 2017-03-02:
@0xc0170 It definitely could be more clearly documented on our platform page. I will take care of that.

I agree that exporting should be disabled so these kinds of issues are avoided. Please let me know what disabling the exporter looks like and I'll be happy to make a PR.

mfiore02, 2017-03-02:
@0xc0170 I added a warning to our platform page recommending against exporting. Should have done that a lot sooner.

0xc0170, 2017-03-02:
> I agree that exporting should be disabled so these kinds of issues are avoided. Please let me know what disabling the exporter looks like and I'll be happy to make a PR.

@theotherjimmy can you please help?

theotherjimmy, 2017-03-02:
I can create a list of "working" post-build steps. These are "working" in the sense that the exporters either flash at an offset (and the target has a soft-device) or they know about the soft-device and can merge it. After that, I would disable exports for all not-supported post-build scripts. This should get you what you want.

theotherjimmy, 2017-03-02:
https://github.com/ARMmbed/mbed-os/pull/3875

theotherjimmy, 2017-03-02:
@0xc0170 @mray19027 @mfiore02  The aforementioned PR should resolve this issue by preventing export for targets we know will fail via post-binary hooks.

mfiore02, 2017-03-03:
LGTM! Thanks @theotherjimmy!

theotherjimmy, 2017-03-03:
I'm quite happy with the implementation, actually. So yeah LGTM!

0xc0170, 2017-03-14:
Since 5.4.1, this export wont be allowed.

closed 2017-03-14 by 0xc0170