ARMmbed/mbed-os-example-blinky #47: [OOB mbed-os-5.4] Compilation failure: uVision export for MTS_MDOT_F411RE , NUCLEO_F411RE

mray19027, 2017-02-28:
## Description
Target: MTS_MDOT_F411RE and NUCLEO_F411RE
Toolchain: export uvision5
mbed-os version: 5.4 OOB
Error: 
```
Object main.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
.\BUILD\mbed-os-example-blinky.axf: Error: L6367E: main.o(i.__sti___8_main_cpp_led1) attributes are not compatible with the provided attributes .
Object timeout.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
.\BUILD\mbed-os-example-blinky.axf: Error: L6367E: timeout.o(.rrx_text) attributes are not compatible with the provided attributes .
Object timeout.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
.\BUILD\mbed-os-example-blinky.axf: Error: L6367E: timeout.o(i._ZN4mbed7Timeout7handlerEv) attributes are not compatible with the provided attributes .
```

0xc0170, 2017-03-02:
The above referenced PR 3861 fixes this issue. Will be fixed in rc2

mbartling, 2017-03-02:
Same issue with efm32pg_stk3401 @theotherjimmy

closed 2017-03-03 by 0xc0170