ARMmbed/mbed-os-example-blinky #62: [OOB 5.5.0] REALTEK_RTL8195AM compiles, but fails to flash in uvision

mray19027, 2017-06-08:
REALTEK_RTL8195AM compiles in uvision, but fails to find the correct target to flash to.

mbed OS: `92fbf2a9b3988d430482fc25a6077f2462e2a634`
mbed CLI version: `1.1.1`

```
".\BUILD\mbed-os-example-blinky.axf" - 0 Error(s), 297 Warning(s).
Build Time Elapsed:  00:01:38
Load "C:\\cygwin64\\home\\micray01\\mbed-os\\mbed-os-example-blinky\\BUILD\\mbed-os-example-blinky.axf" 
No Algorithm found for: 10006000H - 1000FFFFH
No Algorithm found for: 10010000H - 10016F83H
Erase skipped!
Error: Flash Download failed  -  "Cortex-M3"
Flash Load finished at 17:34:35
```

0xc0170, 2017-06-09:
cc @Archcady @theotherjimmy

theotherjimmy, 2017-06-09:
@mray19027 @0xc0170 That's the expected behavior of a "generic" export. The target does not have a CMSIS PACK associated with it.

closed 2017-06-09 by mray19027

0xc0170, 2017-09-27:
> @mray19027 @0xc0170 That's the expected behavior of a "generic" export. The target does not have a CMSIS PACK associated with it.

What should be behaviour for this? Is it expected to work with generic flash algo? shouldn't this when exporting issue a warning that device is missing flash algo, generic one used ?

theotherjimmy, 2017-09-27:
@0xc0170 

>What should be behaviour for this?

That the flash may not work correctly

>Is it expected to work with generic flash algo? 

Nope. It's expected that the user will do the remaining setup.

>shouldn't this when exporting issue a warning that device is missing flash algo, generic one used ?

Probably, but the website will just silently eat the warning and do nothing with it.