ARMmbed/mbed-os-example-blinky #63: [OOB 5.5.0] REALTEK_RTL8195AM export make_gcc flash failure

mray19027, 2017-06-08:
Flashing a binary compiled with the makefile exported from `make_gcc_arm` times out for REALTEK_RTL8195AM 

mbed OS: `92fbf2a9b3988d430482fc25a6077f2462e2a634`
mbed CLI version: `1.1.1`

Command to reproduce:
```
mbed export -i make_gcc_arm
make
```

Fail message:
```
The transfer timed out.
```

Daplink details:
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 46000000061593ce00000000000000000000000097969902
HIC ID: 97969902
Auto Reset: 0
Automation allowed: 0
Daplink Mode: Interface
Interface Version: 0241
Git SHA: e6046fae5e2d3433b04ed95bb80a96375e8db68b
Local Mods: 1
USB Interfaces: MSD, CDC, HID
Interface CRC: 0xb2086ac7
```

sg-, 2017-06-12:
Does this has a post build hook that is needed? It might not be compatible with exporters without extra special tooling.

0xc0170, 2017-06-14:
cc @Archcady

mray19027, 2017-06-16:
@kegilbert Did we get any resolution from this?

0xc0170, 2017-06-19:
> cc @Archcady

Bump

> @kegilbert Did we get any resolution from this?

Any update? I haven't heard yet.

kegilbert, 2017-06-27:
Sorry for the late reply! The REALTEK_RTL8195AM target is not supported in the exporter since https://github.com/ARMmbed/mbed-os/commit/a5cc8ef446a10619bfde0fe9e272a37e56de9213 from my understanding. With the latest master I get the target is not supported error at any rate.

0xc0170, 2017-06-28:
Thanks @kegilbert .IF this is not yet tracked, this should be moved to mbed-os repository as a feature request.

0xc0170, 2017-08-14:
I am closing this as above, not supported currently

closed 2017-08-14 by 0xc0170