ARMmbed/mbed-os-example-blinky #64: make_iar throwing errors after running make

brandonboesch, 2017-06-12:
After exporting to make_iar, and then running make, the following error appears:

[image]

0xc0170, 2017-06-13:
@brandonboesch What is htte platform? is make in your path?

0xc0170, 2017-06-13:
If I run make iar locally, I get this linking error for KW24D:

```
Error[Li009]: runtime model conflict: Module setjmp.o(m7M_tls.a) specifies that
          '__dlib_jmp_buf_num_elements' must be '16', but module unity.o has
          the value '8'
```

brandonboesch, 2017-06-13:
I am running on a FRDM-KW24D, in cmd.exe on windows 10 machine.  I added gnuwin32 to my windows machine and added it to my path variable so that I can call make.  Make works on another example using make_gcc_arm, but I have not tested make_iar with any other examples.  IAR is licensed and working on my machine for compiling as well.

brandonboesch, 2017-06-14:
I resolved this issue.  I had to update my path variable to point to IAR's bin.  Also, the application did not run properly until I updated the KW24D's bootloader firmware to the version found here:

\\venus.cambridge.arm.com\Groups\DevSys\mbed\DAPLink\releases\uvision_0243_2f498feb_kw24d\k20dx_frdmkw24d_if_crc_legacy_0x8000.bin

0xc0170, 2017-06-19:
Thanks for the update. If there is any msg we could do better (like path missing or anything) please create an issue on mbed-os. I'll close this one here

closed 2017-06-19 by 0xc0170