ARMmbed/mbed-os-example-blinky #65: [OOB 5.5.0] NUCLEO_F767ZI gcc makefile binary not working

toyowata, 2017-06-13:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target:**
NUCLEO_F767ZI

**Toolchain:**
GCC_ARM

**Toolchain version:** 
6.3.1 20170215

**mbed-cli version:** 
1.1.1

**Expected behavior**
blinky example should work.

**Actual behavior**
blinky example did not work (LED1 was not blinking).

**Steps to reproduce**
```
$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky
$ mbed update mbed-os-5.5.0-rc1-oob
$ mbed export -m NUCLEO_F767ZI -i make_gcc_arm
$ make
$ copy BUILD/mbed-os-example-blinky.bin f:
```

Build log file: 
[build_log.txt](https://github.com/ARMmbed/mbed-os-example-blinky/files/1070242/build_log.txt)

Generated Makefile:
[Makefile.zip](https://github.com/ARMmbed/mbed-os-example-blinky/files/1070245/Makefile.zip)

**Other notes**

The binary which build by ```mbed compile -m NUCLEO_F767ZI -t GCC_ARM```, works fine.

0xc0170, 2017-06-14:
cc @bcostm @adustm @LMESTM @jeromecoutant

LMESTM, 2017-06-14:
@toyowata @0xc0170 what is mbed-os-5.5.0-rc1-oob ?

0xc0170, 2017-06-14:
release candidate 1 - out of box testing - just use 5.5 branch to retest this, should be able to reproduce

0xc0170, 2017-06-19:
@LMESTM Have you had a chance to reproduce this issue?

MarceloSalazar, 2017-06-28:
@LMESTM could you please comment?

LMESTM, 2017-06-29:
@MarceloSalazar @0xc0170 @toyowata 
Indeed issue was reproduced. Quick debugging was showing similar behavior to some of the issues we've debugged in MBED after move to CMSIS5. 
As several of those errors were solved in the past 2 weeks and were not related to STM targets specifically, I tried a more recent version 
```
$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky
$ mbed update mbed-os-5.5.1
$ mbed export -m NUCLEO_F767ZI -i make_gcc_arm
$ make
$ copy BUILD/mbed-os-example-blinky.bin f:
```
And this works ! 
So I'd suggest to close this issue ...

closed 2017-06-29 by MarceloSalazar