ARMmbed/mbed-os-example-blinky #81: mbed-os.lib not pointing to mbed 5.6.0 release candidate in branch mbed-os-5.6.0-oob

dhwalters423, 2017-09-13:
It appears mbed-os.lib is not pointing to the correct commit number that is tagged as the mbed OS 5.6.0 release candidate in the branch mbed-os-5.6.0-oob. This is causing the CLI and mbed-ls tools to believe it has checked out the correct branch when running ``mbed update mbed-os-5.6.0-oob`` but the underlying mbed-os folder is still reporting as being checked out to mbed-os-5.5.0.

For example, when checking out the mbed-os-5.6.0-oob branch:
```
C:\Users\davwal04\Desktop\Out of box campaigns\mbed OS 5.6\mbed-os-example-blinky>mbed update mbed-os-5.6.0-oob
[mbed] Updating program "mbed-os-example-blinky" to branch mbed-os-5.6.0-oob
[mbed] Updating library "mbed-os" to rev #afffea9a848b
fatal: reference is not a tree: afffea9a848b443796d952cb406206f1a6f8d257
[mbed] ERROR: Unable to update "mbed-os" to rev #afffea9a848b
```

When looking inside the mbed-os-example-blinky folder:
```
C:\Users\davwal04\Desktop\Out of box campaigns\mbed OS 5.6\mbed-os-example-blinky>git status
On branch mbed-os-5.6.0-oob
Your branch is up-to-date with 'origin/mbed-os-5.6.0-oob'.
nothing to commit, working tree clean
```

but when you look into the underlying mbed-os folder:

```
C:\Users\davwal04\Desktop\Out of box campaigns\mbed OS 5.6\mbed-os-example-blinky>cd mbed-os
C:\Users\davwal04\Desktop\Out of box campaigns\mbed OS 5.6\mbed-os-example-blinky\mbed-os>git status
On branch mbed-os-5.5
Your branch is up-to-date with 'origin/mbed-os-5.5'.
nothing to commit, working tree clean
```

The correct commit that mbed-os.lib should be pointing to is:
+https://github.com/ARMmbed/mbed-os/#a4056fb8bf3ee10b8e037d332f8fbd0c68169ef1

Please reference PR: https://github.com/ARMmbed/mbed-os-example-blinky/pull/80/commits/6f8c4e64586836b457164183fdfc364b63bcb631

MarceloSalazar, 2017-09-13:
@dhwalters423 thanks for raising this! I'm also able to reproduce

@sg- @adbridge are you ok if we update all examples to suggested commit? (a4056fb8bf3ee10b8e037d332f8fbd0c68169ef1)

MarceloSalazar, 2017-09-13:
@dhwalters423 fyi @adbridge has re-created the branches using this name: mbed-os-5.6-oob
I've verified and works ok. OOB spreadsheet updated.

closed 2017-09-13 by MarceloSalazar