ARMmbed/mbed-os-example-blinky #83: Compilation failure with IAR 8 with NUCLEO_F429ZI target

vijhar01, 2017-09-14:
**PS C:\Users\vijhar01\mbed-os-example-blinky> mbed detect**

[mbed] Detected NUCLEO_F429ZI, port COM6, mounted D:
[mbed] Supported toolchains for NUCLEO_F429ZI
+---------------+-----------+-----------+-----------+-----------+-----------+
| Target        | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |
+---------------+-----------+-----------+-----------+-----------+-----------+
| NUCLEO_F429ZI | Supported | Supported | Supported | Supported | Supported |
+---------------+-----------+-----------+-----------+-----------+-----------+
Supported targets: 1
Supported toolchains: 3

**PS C:\Users\vijhar01\mbed-os-example-blinky> mbed compile -t IAR -m NUCLEO_F429ZI**
Building project mbed-os-example-blinky (NUCLEO_F429ZI, IAR)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Compile [  0.2%]: cmain.S
Compile [  0.5%]: AnalogIn.cpp
Compile [  0.7%]: BusIn.cpp
Compile [  1.0%]: BusInOut.cpp
Compile [  1.2%]: main.cpp
Compile [  1.5%]: Ethernet.cpp
Compile [  1.7%]: BusOut.cpp
Compile [  2.0%]: CAN.cpp
Compile [  2.2%]: FlashIAP.cpp
Compile [  2.5%]: I2CSlave.cpp
Compile [  2.7%]: I2C.cpp
Compile [  3.0%]: InterruptIn.cpp
Compile [  3.2%]: InterruptManager.cpp
Compile [  3.5%]: RawSerial.cpp
Compile [  3.7%]: SPISlave.cpp
Compile [  4.0%]: SPI.cpp
Compile [  4.2%]: Serial.cpp
Compile [  4.5%]: SerialBase.cpp
Compile [  4.7%]: Ticker.cpp
Compile [  5.0%]: Timeout.cpp
Compile [  5.2%]: TimerEvent.cpp
Compile [  5.5%]: Timer.cpp
Compile [  5.7%]: UARTSerial.cpp
Compile [  6.0%]: equeue.c
Compile [  6.2%]: EventQueue.cpp
Compile [  6.5%]: equeue_mbed.cpp
Compile [  6.7%]: emac_lwip.c
Compile [  7.0%]: emac_stack_lwip.cpp
Compile [  7.2%]: equeue_posix.c
Compile [  7.5%]: stm32f4_eth_conf.c
Compile [  7.7%]: stm32f4_eth_init.c
Compile [  8.0%]: lwip_checksum.c
Compile [  8.2%]: lwip_memcpy.c
Compile [  8.5%]: EthernetInterface.cpp
Compile [  8.7%]: stm32xx_emac.c
[Warning] stm32xx_emac.c@98,24: [Pe177]: variable "regvalue" was declared but never referenced
[Warning] stm32xx_emac.c@99,0: [Pe550]: variable "hal_eth_init_status" was set but never used
Compile [  9.0%]: lwip_random.c
Compile [  9.2%]: lwip_sys_arch.c
Compile [  9.5%]: lwip_tcp_isn.c
Compile [  9.7%]: lwip_api_lib.c
Compile [ 10.0%]: lwip_err.c
Compile [ 10.2%]: lwip_api_msg.c
Compile [ 10.5%]: lwip_netdb.c
Compile [ 10.7%]: lwip_netifapi.c
Compile [ 11.0%]: lwip_netbuf.c
Compile [ 11.2%]: lwip_sockets.c
Compile [ 11.5%]: lwip_autoip.c
Compile [ 11.7%]: lwip_tcpip.c
Compile [ 12.0%]: lwip_icmp.c
Compile [ 12.2%]: lwip_etharp.c
Compile [ 12.5%]: lwip_dhcp.c
[... 261 lines cut ...]

          _file->_Rend = _file->_Wend;
               ^
"C:\Users\vijhar01\mbed-os-example-blinky\mbed-os\platform\mbed_retarget.cpp",866  Error[Pe393]: pointer to incomplete c
lass type is not allowed

          _file->_Rend = _file->_Wend;
                              ^
"C:\Users\vijhar01\mbed-os-example-blinky\mbed-os\platform\mbed_retarget.cpp",866  Error[Pe393]: pointer to incomplete c
lass type is not allowed

          _file->_Next = _file->_Wend;
               ^
"C:\Users\vijhar01\mbed-os-example-blinky\mbed-os\platform\mbed_retarget.cpp",867  Error[Pe393]: pointer to incomplete c
lass type is not allowed

          _file->_Next = _file->_Wend;
                              ^
"C:\Users\vijhar01\mbed-os-example-blinky\mbed-os\platform\mbed_retarget.cpp",867  Error[Pe393]: pointer to incomplete c
lass type is not allowed

[mbed] ERROR: "C:\Python27\python.exe" returned error code 1.
[mbed] ERROR: Command "C:\Python27\python.exe -u C:\Users\vijhar01\mbed-os-example-blinky\mbed-os\tools\make.py -t IAR -
m NUCLEO_F429ZI --source . --build .\BUILD\NUCLEO_F429ZI\IAR" in "C:\Users\vijhar01\mbed-os-example-blinky"
---

MarceloSalazar, 2017-09-15:
cc @adbridge

adbridge, 2017-09-15:
Strange this build in the pr for IAR 8 - 
```
Compile [ 64.9%]: mbed_retarget.cpp
[DEBUG] Compile: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.0\arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 -Oh --cpu Cortex-M4F --thumb --dlib_config DLib_Config_Full.h --c++ --no_rtti --no_exceptions --guard_calls --no_static_destruction -DTARGET_STM32F429xx -DDEVICE_SPISLAVE=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DUSBHOST_OTHER -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_NUCLEO_F429ZI -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DTARGET_STM32F4 -D__CMSIS_RTOS -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_PORTOUT=1 -DDEVICE_SPI_ASYNCH=1 -DTOOLCHAIN_IAR -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTARGET_STM32F429 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_PWMOUT=1 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DDEVICE_TRNG=1 -DTARGET_LIKE_MBED -DTARGET_STM32F429ZI -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DUSB_STM_HAL -DTARGET_STM32F429xI -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DARM_MATH_CM4 -DMBED_BUILD_TIMESTAMP=1505080586.76 -f .\BUILD\NUCLEO_F429ZI\IAR\.includes_82eefb55eda86348fecbc5857b0d13b6.txt --preinclude=.\BUILD\NUCLEO_F429ZI\IAR\mbed_config.h --dependencies .\BUILD\NUCLEO_F429ZI\IAR\mbed-os\platform\mbed_retarget.d -l .\BUILD\NUCLEO_F429ZI\IAR\mbed-os\platform\mbed_retarget.s.txt -o .\BUILD\NUCLEO_F429ZI\IAR\mbed-os\platform\mbed_retarget.o .\mbed-os\platform\mbed_retarget.cpp
[DEBUG] Return: 0
Compile [ 65.1%]: RTX_Config.c
```

adbridge, 2017-09-15:
@vijhar01  What platform did you build on and what exact version of IAR did you use?

adbridge, 2017-09-15:
I have just reproduced this using the windows version of IAR workbench - 
IAR ANSI C/C++ Compiler V8.11.3.13950/W32 for ARM
Running in gitbash on windows 10.

@shrikant I assume we are running the linux version in the ci ?

vijhar01, 2017-09-15:
Hi Anna,

Here are my test environment details:

Build environment – Windows 10
IAR version - 8.11.2.13589 (8.11.2.13589)

Thanks, Viji

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

adbridge, 2017-09-15:
I've built with latest mbed-os master and that builds ok!

adbridge, 2017-09-15:
mmm ok so I now did this - 
> rm -rf mbed-os/                                                                                                                                                                            
> mbed deploy                                                                                                                                                                                
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #a4056fb8bf3e                                                                                               
> mbed compile -t IAR -m NUCLEO_F429ZI            

and this worked for me. 
The first time I tried it I did not remove the previous mbed-os. I'm wondering if you had leftover artifacts ?

@vijhar01 Could you please re-try this but following the above steps ?

vijhar01, 2017-09-18:
That worked.
I was able to compile successfully with IAR

Documentation enhancement:

Publish examples for export into iar 8 workspace.

I ran the following and did not know where to look for the exported iar project

C:\mbed-os-example-blinky> mbed export -m NUCLEO_F429ZI -i IAR
Scan: .
Scan: FEATURE_LWIP
PS C:\mbed-os-example-blinky>

I found an mbed-os-example-blinky.eww file in the current directory and was able to open it in the IAR workbench. I hope that is the right file.

Thanks, Viji

MarceloSalazar, 2017-09-19:
> Documentation enhancement:
> Publish examples for export into iar 8 workspace.
> I ran the following and did not know where to look for the exported iar project

@AnotherButler FYI find here some feedback on docs for exporters

AnotherButler, 2017-09-19:
Thanks for letting me know @MarceloSalazar I've created a JIRA issue for this.

closed 2017-09-20 by vijhar01