ARMmbed/mbed-os-example-blinky #86: mbed export fails

LMESTM, 2017-09-26:
#description

I imported this example in my environment 
`mbed` import `mbed-os-example-blinky`

Compiled ok:
` mbed compile -t IAR -m NUCLEO_L476RG `

Trying export fails:
```
$ mbed export -i IAR -m NUCLEO_L476RG  -c -vv
Traceback (most recent call last):
  File "C:\Data\Workspace\mbed-os-example-blinky\mbed-os\tools\project.py", line 256, in <module>
    main()
  File "C:\Data\Workspace\mbed-os-example-blinky\mbed-os\tools\project.py", line 248, in main
    rmtree(BUILD_DIR)
  File "c:\python27\lib\shutil.py", line 239, in rmtree
    onerror(os.listdir, path, sys.exc_info())
  File "c:\python27\lib\shutil.py", line 237, in rmtree
    names = os.listdir(path)
WindowsError: [Error 3] The system cannot find the path specified: 'C:\\Data\\Workspace\\mbed-os-example-blinky\\mbed-os\\BUILD/*.*'
[mbed] Working path "C:\Data\Workspace\mbed-os-example-blinky" (program)
[mbed] Exec "c:\python27\python.exe -u C:\Data\Workspace\mbed-os-example-blinky\mbed-os\tools\project.py -i iar -m NUCLEO_L476RG -c --source ." in C:\Data\Workspace\mbed-os-example-blinky
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Data\Workspace\mbed-os-example-blinky\mbed-os\tools\project.py -i iar -m NUCLEO_L476RG -c --source ." in "C:\Data\Workspace\mbed-os-example-blinky"
---
```

Windows OS, with MINGW64

0xc0170, 2017-09-26:
I reproduced the issue. The traceback is caused by `-c` argument , and build is not there. This should be handled normally.

cc @theotherjimmy

theotherjimmy, 2017-09-26:
Agreed. Let me see if this is cross platform.

theotherjimmy, 2017-09-26:
Nope. Seems like it's a Windows only issue. Sorry.

theotherjimmy, 2017-09-28:
@LMESTM Could you open an issue for this on mbed-os? This is a problem with all `mbed export -c` which is part of the Mbed OS tools.

0xc0170, 2017-10-09:
> @LMESTM Could you open an issue for this on mbed-os? This is a problem with all mbed export -c which is part of the Mbed OS tools.

@LMESTM bump

LMESTM, 2017-10-09:
Done: 
https://github.com/ARMmbed/mbed-os/issues/5276

closed 2017-10-09 by 0xc0170