ARMmbed/mbed-os-example-blinky #92: [RTL8195AM-Ameba]: On-board LED's not mapped? Blinky not working.

ashok-rao, 2017-11-01:
@samchuarm , I was trying to run blinky on RTL8195AM-Ameba [DEV01_1V0] target and I could not find the mapping for the on-board LED's to work with blinky. I found from the pin mapping here: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/PinNames.h#L200  that LED's are mapped to the Arduino headers?
Can you please advice about on-board LED pin maps? Thanks.

ashok-rao, 2017-11-01:
@Archcady .. could you please let me know if on-board LED's are indeed mapped out to be USER LED's or not? I also tried physically connecting an LED between D8 on the Arduino header and GND, and it still seems to be not working. I'm using Mbed OS 5.6.3. Thanks.

Archcady, 2017-11-02:
The on-board LED's are reversed for other usages, users need to connect GPIO with their own LEDs.

ashok-rao, 2017-11-02:
Wiring the LED between D8 and Vcc (3.3V) on the Arduino headers works with blinky! Thank you. Closing this issue now.
Thanks @Archcady and @samchuarm

closed 2017-11-02 by ashok-rao