ARMmbed/mbed-os-example-blinky #96: Blinky example should print something

JanneKiiskila, 2017-12-01:
As there are multiple boards that do not actually have an user controllable LED, those should just print "Blink1" or something - or actually all boards could print something, say every 10 times it's blinked the (imaginary) LED.

0xc0170, 2017-12-01:
Can you list those boards please?

JanneKiiskila, 2017-12-02:
Well, try out the REALTEK_RTL8195AM for example.

It has plenty of LEDs, just none of the seem to be available for the "user" side.

JanneKiiskila, 2017-12-03:
PR created; https://github.com/ARMmbed/mbed-os-example-blinky/pull/97

JanneKiiskila, 2017-12-04:
Raised also an issue towards Realtek; https://github.com/ARMmbed/mbed-os/issues/5642

0xc0170, 2018-03-28:
Realtek fixed their documentation, this should be resolved. This example blinks only

Thanks for the report for realtek

closed 2018-03-28 by 0xc0170