ARMmbed/mbed-os-example-blinky #99: Misleading output about VK_RZ_A1H in 5.7

MarceloSalazar, 2017-12-06:
Branch: mbed-os-5.7.0-oob

It seems VK_RZ_A1H (Cortex-A) is not supported.

We should update the message "**Will be supported in mbed OS 5.6**"  with something like "Will be supported in a future version of Mbed OS"

@0xc0170

```
mac:~/mbed/OOB/5.7/mbed-os-example-blinky$ mbed compile -t GCC_ARM -m VK_RZ_A1H
Building project mbed-os-example-blinky (VK_RZ_A1H, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP

Could not compile for VK_RZ_A1H: VK_RZ_A1H Will be supported in mbed OS 5.6. To use the VK_RZ_A1H, please checkout the mbed OS 5.4 release branch. See https://developer.mbed.org/platforms/Renesas-GR-PEACH/#important-notice for more information
```

toyowata, 2017-12-06:
@MarceloSalazar 
The PR is still in review here:
https://github.com/ARMmbed/mbed-os/pull/5628

0xc0170, 2017-12-06:
Thanks @toyowata . 

We should update the message "Will be supported in mbed OS 5.6" with something like "Will be supported in a future version of Mbed OS"

The message comes from mbed-os, it should be requested there, it's not part of this example.

MarceloSalazar, 2017-12-06:
@theotherjimmy please have a look. We'll appreciate if you could patch this on Mbed OS.

closed 2017-12-12 by adbridge