ARMmbed/mbed-os-example-blinky #102: Online compiler error for the Realtek RTL8195AM board

gggekov, 2017-12-08:
I'm unable to compile blinky in the online compiler for the Realtek RTL8195AM board. I'm on the `mbed-os-5.7.0-oob` branch.  I'm getting Error L6236E. See the screenshot for the complete error message
[image]

Can you please fix this issue?

Regards,
George

MarceloSalazar, 2017-12-08:
@theotherjimmy

theotherjimmy, 2017-12-08:
I have only seen this happen when the linker garbage collected _ALL_ code. The fact that you have a linker script implies that you have mbed-os, so I don't know what else to try here.

gggekov, 2017-12-08:
It compiles fine with ARM in mbed cli. There should be a way to compile it correctly in the online IDE as well.

adbridge, 2017-12-13:
@theotherjimmy what is the status of this ? Do we think it is a real problem in the tools that may need to be a known issue ?

theotherjimmy, 2017-12-13:
@adbridge The online compiler adds the target name between the basename and the extension of the file, that is:

 line-edness | Source | Obj. file
-------------|--------|------------
offline | `mbed_retarget.cpp` | `mbed_retarget.o`
online | `mbed_retarget.cpp` | `mbed_retarget.RTL8195AM.o`

The Realtek scatter file includes a match for a specific object file with the format `*<filename>.o`, and will not match in the online compiler. This is where the link to main is contained, and a lack of a link to main allows the linker to garbage collect all of the user program and mbed OS. Then you get a message about no sections first or last.

This target has not ever worked in the online compiler due to this issue. We could change the workings of the scatter file or we could adopt a naming scheme for object files that matches the naming scheme offline.

samchuarm, 2017-12-14:
@ARMmbed/team-realtek

adbridge, 2017-12-14:
@theotherjimmy @samchuarm If this has never worked in the online compiler it shouldn't even have mbed enabled status.... This should be removed until it is fixed

tung7970, 2017-12-15:
@theotherjimmy @ARMmbed/team-realtek Hi, the quick fix is to rewrite the match condition from 

*<filename>.o to *<filename>*.o
e.g.
*rtl8195a_init.o (+RO) => *rtl8195a_init*.o (+RO)

Then, the scatter file should works for both online and offline compilers.

yangkunming99, 2017-12-15:
@tung7970, thanks for the comment. Is there any way we can force FIRST in c file instead of scatter file?

Archcady, 2017-12-15:
Hello @samchuarm can I send you the changes for the online compiler once and when you confirm it is working, ill raise a PR immediately?

tung7970, 2017-12-15:
@yangkunming99 Unfortunately, placement is determined by linker.

0xc0170, 2018-03-28:
Hi, this should be resolved via referenced PR above (see mbed-os issue 5626)

closed 2018-03-28 by 0xc0170