ARMmbed/mbed-os-example-blinky #104: Exporters not supported for Realtek RTL8195AM

gggekov, 2017-12-08:
I'm not able to export the blinky project for the Realtek Ameba board. 
I'm using Windows10 and I'm getting 
`C:\Users\geogek01\Documents\mbedCLIproj\mbed-os-example-blinky [mbed-os-5.7.0-oob ≡ +3 ~1 -0 !]> mbed export -i make_iar -m realtek_
rtl8195am
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S [{matrix,ides}]] [-E] [--source SOURCE_DIR]
                  [-D MACROS] [--profile PROFILE] [--update-packs]
                  [--app-config APP_CONFIG]
project.py: error: REALTEK_RTL8195AM not supported by make_iar`

I am getting the same error message for `make_armc5` and for `make_gcc_arm`

I'm using mbed-cli 1.1.1

Can you also take a look into this issue please?

Thanks,
George

MarceloSalazar, 2017-12-08:
@theotherjimmy another one ;-)

theotherjimmy, 2017-12-08:
@gggekov The Realtek board has a post build script that the tools cannot replicate. @MarceloSalazar Please stop "testing" these not supported combinations.

MarceloSalazar, 2017-12-08:
@theotherjimmy sorry we were not aware of this issue (not in the known-issue list).

@adbridge can we add this to the known issue list please?

theotherjimmy, 2017-12-08:
This is not a known issue. It's not supported. @MarceloSalazar This is what the support matrix looks like for exporters:
```
+-----------------------+-------------+-------+-------+----------+---------------+-----------------+-------------+--------+---------+---------------+-----+-----+------------+------------+------------+--------------+----------+------------+-----------+--------------+----------+---------+----------+--------------+----------------+------------+
| Platform              | atmelstudio | coide | ds5_5 | e2studio | eclipse_armc5 | eclipse_gcc_arm | eclipse_iar | embitz | gcc_arm | gnuarmeclipse | iar | kds | lpcxpresso | make_armc5 | make_armc6 | make_gcc_arm | make_iar | mcuxpresso | qtcreator | simplicityv3 | sw4stm32 | uvision | uvision5 | vscode_armc5 | vscode_gcc_arm | vscode_iar |
+-----------------------+-------------+-------+-------+----------+---------------+-----------------+-------------+--------+---------+---------------+-----+-----+------------+------------+------------+--------------+----------+------------+-----------+--------------+----------+---------+----------+--------------+----------------+------------+
| ARCH_BLE              |      -      |   -   |   -   |    x     |       x       |        x        |      -      |   -    |    x    |       x       |  -  |  -  |     -      |     x      |     -      |      x       |    -     |     -      |     x     |      -       |    x     |    x    |    x     |      x       |       x        |     -      |
+-----------------------+-------------+-------+-------+----------+---------------+-----------------+-------------+--------+---------+---------------+-----+-----+------------+------------+------------+--------------+----------+------------+-----------+--------------+----------+---------+----------+--------------+----------------+------------+
| ARCH_BLE_BOOT         |      -      |   -   |   -   |    x     |       x       |        x        |      -      |   -    |    x    |       x       |  -  |  -  |     -      |     x      |     -      |      x       |    -     |     -      |     x     |      -       |    x     |    x    |    x     |      x       |       x        |     -      |
+-----------------------+-------------+-------+-------+----------+---------------+-----------------+-------------+--------+---------+---------------+-----+-----+------------+------------+------------+--------------+----------+------------+-----------+--------------+----------+---------+----------+--------------+----------------+------------+
| ARCH_BLE_OTA          |      -      |   -   |   -   |    x     |       x       |        x        |      -      |   -    |    x    |       x       |  -  |  -  |     -      |     x      |     -      |      x       |    -     |     -      |     x     |      -       |    x     |    x    |    x     |      x       |       x        |     -      |
+-----------------------+-------------+-------+-------+----------+---------------+-----------------+-------------+--------+---------+---------------+-----+-----+------------+------------+------------+--------------+----------+------------+-----------+--------------+----------+---------+----------+--------------+----------------+------------+
| ARCH_GPRS             |      -      |   -   |   -   |    x     |       x       |        x        |      x      |   x    |    x    |       x       |  -  |  -  |     -      |     x      |     -      |      x       |    x     |     -      |     x     |      -       |    x     |    x    |    x     |      x       |       x        |     x      |
+-----------------------+-------------+-------+-------+----------+---------------+-----------------+-------------+--------+---------+---------------+-----+-----+------------+------------+------------+--------------+----------+------------+-----------+--------------+----------+---------+----------+--------------+----------------+------------+
| ARCH_LINK             |      -      |   -   |   -   |    x     |       x       |        x        |      -      |   -    |    x    |       x       |  -  |  -  |     -      |     x      |     -      |      x       |    -     |     -      |     x     |      -       |    x     |    x    |    x     |      x       |       x        |     -      |
+-----------------------+-------------+-------+-------+----------+---------------+-----------------+-------------+--------+---------+---------------+-----+-----+------------+------------+------------+--------------+----------+------------+-----------+--------------+----------+---------+----------+--------------+----------------+------------+
| ARCH_LINK_BOOT        |      -      |   -   |   -   |    x     |       x       |        x        |      -      |   -    |    x    |       x       |  -  |  -  |     -      |     x      |     -      |      x       |    -     |     -      |     x     |      -       |    x     |    x    |    x     |      x       |       x        |     -      |
+-----------------------+-------------+-------+-------+----------+---------------+-----------------+-------------+--------+---------+---------------+-----+-----+------------+------------+------------+--------------+----------+------------+-----------+--------------+----------+---------+----------+--------------+----------------+------------+
| ARCH_LINK_OTA         |      -      |   -   |   -   |    x     |       x       |        x        |      -      |   -    |    x    |       x       |  -  |  -  |     -      |     x      |     -      |      x       | [... cut ...] -------------+----------------+------------+
Total IDEs: 26
Total platforms: 232
Total permutations: 3577
```
 Do you want to add the fact that none of the ARCH boards (except the ARCH_PRO) don't support IAR to the list of know issues as well, or how about that MCUXpresso supports only NXP products (and then only some of them)? I don't think that adding all of this information to the known issues list is a good idea, as it would drown out all actual issues, and make the list practically useless.

MarceloSalazar, 2017-12-08:
We aim to add more information over time about platform support and issues with mbed tools.
Hiding issues don't help to provide a good UX.

@adbridge let's discuss next week to find the best way to document this.

theotherjimmy, 2017-12-08:
@MarceloSalazar I don't see how this is being hidden. The error message was very clean about what went wrong: The target is not supported by the exporter.

adbridge, 2017-12-11:
Maybe this information should be in the exporters section of the online documentation ? I.e. the matrix of support.....

adbridge, 2017-12-12:
@AnotherButler WRT my comment above - thoughts ?

AnotherButler, 2017-12-12:
I would have thought this content would be a better fit for the platforms page that lists all of the boards: https://os.mbed.com/platforms/

MarceloSalazar, 2017-12-15:
Internal requirement: IOTPREQ-2437

0xc0170, 2017-12-20:
Referenced isse on mbed OS as tracking issue, this can be closed.

closed 2017-12-20 by 0xc0170