ARMmbed/mbed-os-example-blinky #105: This GettingStarted.html generated by "mbed export" redirect to a broken documentation page

jamesbeyond, 2017-12-11:
I followed the guide to run "mbed export -m K64F -i uvision" command.
the mbed cli created a "GettingStarted.html" in my local working directory.
By double click on the HTML file, this redirect me to "https://os.mbed.com/docs/v5.6/tools/exporting.html/"
Which is not a very well formatted documentation page, and all the URL links on that page had broken

MarceloSalazar, 2017-12-11:
@theotherjimmy please have a look

0xc0170, 2017-12-11:
@AnotherButler I can fix this if I know where it should point it (did the link changed) ?

AnotherButler, 2017-12-11:
@0xc0170 The problem with that link is the slash at the end. https://os.mbed.com/docs/v5.6/tools/exporting.html formats correctly. If you delete the slash, I think it should be OK.

0xc0170, 2017-12-11:
Thanks, sending PR now :)

0xc0170, 2017-12-20:
This was fixed

closed 2017-12-20 by 0xc0170