ARMmbed/mbed-os-example-blinky #118: Compile problem

drvrh, 2018-04-04:
Hello
when I compile mbed-os-example-blinky 

`mbed compile -t GCC_ARM -m NUCLEO_L152RE -c `

get error:

```
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS]
               [-S [{matrix,toolchains,targets}]] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rpc] [--usb] [--dsp]
               [--testlib] [--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: Could not find executable for GCC_ARM.
Currently set search path: ~/opt/gcc-arm-none-eabi-6-2017-q2-update/bin/
[mbed] ERROR: "/usr/bin/python" returned error code 2.
[mbed] ERROR: Command "/usr/bin/python -u /home/grega/podatki/seltron/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM" in "/home/grega/podatki/seltron/mbed-os-example-blinky"
---
```

My mbed config --list look like this:

```
[mbed] Global config:
MBED_GCC_ARM_PATH=~/opt/gcc-arm-none-eabi-6-2017-q2-update/bin/

[mbed] Local config (/home/grega/podatki/seltron/mbed-os-example-blinky):
No local configuration is set
```
In mbed_settings.py I set:
# GCC ARM
GCC_ARM_PATH = "~/opt/gcc-arm-none-eabi-6-2017-q2-update/bin/"

Where is the problem?

0xc0170, 2018-04-06:
Can you run it manually with the defined path ? For instance run `arm-none-eabi-gcc --version` command?

0xc0170, 2018-04-11:
IF this problem persists, please ask in mbed OS forum or in mbed OS github

closed 2018-04-11 by 0xc0170

drvrh, 2018-04-11:
Sorry I forgot answer. 
Problem in global PATH. When I added new environment in gloabl path (debian) then works.