ARMmbed/mbed-os-example-blinky #124: [Mbed-OS 5.9 OOB] RZ_A1H doesn't work with ARM build

toyowata, 2018-06-05:
## Bug

**Target**
GR-PEACH and GR-LYCHEE

**Toolchain:**
ARM

**Expected behavior**
The build binary works on the GR-PEACH/LYCHEE target.

**Actual behavior**
No LED blinky on the target.  No crash log reported.

**Steps to reproduce**
```
$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky
$ mbed update mbed-os-5.9.0-oob
$ mbed compile -m RZ_A1H -t ARM
```

**Version hash**
```
mbed-os-example-blinky (#fe1c51632dff, tag: mbed-os-5.9.0-rc1)
`- mbed-os (#6817dc43f9f5, tag: mbed-os-5.9.0-rc1)
```

**Other information**
* Other build (GCC_ARM and IAR) works fine.
* Last release version (mbed-os-5.8.5 or latest release tag) works fine.
* Renesas also reported this [here](https://github.com/ARMmbed/mbed-os/issues/7065)

cc @MarceloSalazar

MarceloSalazar, 2018-06-05:
FYI @adbridge

toyowata, 2018-06-06:
```--profile=debug``` build prevents the issue.  Its may be optimization related problem.

0xc0170, 2018-06-12:
Thanks for reporting. This is a duplicate, already referenced above . I'll close this

closed 2018-06-12 by 0xc0170