ARMmbed/mbed-os-example-blinky #132: This example does not work after exporting on Keil µVision 5

Navis-Raven, 2018-08-10:
The example works well on Arm mbed Online Compiler but after exporting on Keil and building, it says at least two errors:

1) /filer/workspace_data/exports/8/8bbbb9b5d0fe7814b65b507b54bb7d92/mbed-os-example-blinky/mbed_config.h contains an incorrect path

[image]

2) After building it make at least one error which is underlined in #include "mbed.h". It says: error in include chain (mbed_assert.h): invalid token at start of a preprocessor expression

[image]

Navis-Raven, 2018-08-14:
It is the same issue on nRF52-DK as well as on STM32 Discovery kit IoT node. So we can't use Keil for importation of Mbed projects : This is a major issue !

[image]

cmonr, 2018-08-14:
@Navis-Raven Have you tried using mbed-cli to export mbed-os-example-blinky to uVision 5?
It's strange that this isn't working for you considering we explicitly test exporting examples to uVision 5.

Navis-Raven, 2018-08-14:
I have a similar issue while trying to use it with IAR IDE for Arm 8.30.1 (and for STM32 Discovery kit IoT node). Similar issue, it does not compile, and we have also 145 errors. This issue is quite annoying we cannot go anywhere else with this (changing the hardware or the IDE does not change anything). (I'm on Windows 7) 

[image]

Navis-Raven, 2018-08-14:
@cmonr 
I didn't unederstand what mbed-cli is used for, and its usage is more complex. Adding to that the installation of mbed-cli crashed at the end of it, so, I decided to uninstall it. 
 
I opened this issue here about mbed-cli installation problem
https://github.com/ARMmbed/mbed-cli/issues/733

Navis-Raven, 2018-08-15:
It does not work on VSCode (ARM-GCC) either but not for the same reason, I have to open a new issue. 

At this time I have no IDE option to run mbed API for any of these two hardware (nRF52-DK and STM32 Discovery kit IoT node)

cmonr, 2018-08-15:
@Navis-Raven Would you be able to check if this works with earlier releases and/or different targets?

Navis-Raven, 2018-08-15:
I'm working only with earlier releases, because I've just installed these tools last week in order to try them. 

As you can see I've tried with two different targets and I have similar issues. I have no more Arm based Target.

Navis-Raven, 2018-08-24:
Some additionnal info Keil µVision 5 with nrF52DK
[image]

0xc0170, 2018-08-27:
Hi, I will review the 2 issues regarding exporters, and will report back the status.

theotherjimmy, 2018-09-07:
@Navis-Raven The workaround is to change the path of "mbed_config.h" to be in the root of your project. I'm looking at why you're getting the filer path within your project.

yennster, 2018-09-21:
Hey @theotherjimmy didn't we talk about this issue with the online compiler the other day?

yennster, 2018-09-21:
@theotherjimmy Here's the GitHub issue we were talking about: https://github.com/ARMmbed/mbed-os/issues/7866

ConradBraam, 2018-09-24:
I am pretty sure that I'm watching this git issue only because I raised a jira on it, just recently. Is this best done in jira?

EngSocProject, 2018-09-29:
[image]

Remove the * to fix assert error. I am not sure if thats the offical solultion but its no longer showing that error.

theotherjimmy, 2018-10-01:
@EngSocProject Yes that's strange. I'm not sure either what the problem even is there. The header file is just trying to construct a version number. Further, if you're using uVision, you're probably using Arm Compiler 5. That section looks like it's setup for GCC.

ConradBraam, 2018-10-01:
Hi @EngSocProject , since I've also seen similar issues, I can "try" help.
1. You want to go into linker options, and turn on "Use memory layout from target dialog", and make sure you have debugging info turned off.
But you will get stuck again on running out of flash/RAM anyway, so I suspect there is an issue with the exporter, and if already familiar with Keil, you might do better mileage creating a project virtually from scratch. (try run > mbed export -m LPC1768 -i uvision5 --profile RELEASE)

2. What worked for me was using the commandline however:
c:\mbed-os-example-blinky-oob> mbed compile -m LPC1768 -c -t GCC_ARM
...
| Module           |         .text |       .data |        .bss |
|------------------|---------------|-------------|-------------|
| [fill]           |     615(+615) |       4(+4) |     12(+12) |
| [lib]\c.a        | 18671(+18671) | 2472(+2472) |     89(+89) |
| [lib]\gcc.a      |   3864(+3864) |       0(+0) |       0(+0) |
| [lib]\misc       |     208(+208) |     12(+12) |     28(+28) |
| main.o           |     198(+198) |       4(+4) |     24(+24) |
| mbed-os\drivers  |       56(+56) |       0(+0) |       0(+0) |
| mbed-os\hal      |   2005(+2005) |       4(+4) |     68(+68) |
| mbed-os\platform |   2903(+2903) |   260(+260) |   133(+133) |
| mbed-os\rtos     |   8312(+8312) |   168(+168) | 6053(+6053) |
| mbed-os\targets  |   3538(+3538) |       4(+4) |   241(+241) |
| Subtotals        | 40370(+40370) | 2928(+2928) | 6648(+6648) |
Total Static RAM memory (data + bss): 9576(+9576) bytes
Total Flash memory (text + data): 43298(+43298) bytes

I understand you will want to use Keil and the debugger, but hope this gives you some route forward. Core of the matter is the online/offline exporter is problematic on some targets, for now export to targets I have in my box of targets here work, so mileage will vary.

Kermit66, 2018-10-20:
...same problems...
in main.cpp errore in include chain (mbed_assert.h): invalid token at start of a preprocessor expression
in mbed_assert.h line 100 and 103 error invalid token at start of a preprocessor expression
[image]
[image]

dorduran, 2019-01-06:
Similar issue also appears on "Eclipse-GCC-ARM" exports as well. The build ends up with;
"/mbed_config.h: No such file or directory"

Looks like mbed's export process has the issue. Aperrently mbed Continuous Integration framework does not check if the export process generates an error/warning free build.

zhiyong80, 2019-03-03:
Just ran into this issue as well. The weird thing is, two of my newly exported projects were working totally fine, then all of a sudden, after diner, when I resumed work, both of them started to have this issue.

ConradBraam, 2019-03-04:
The original issue was that the exported Keil project file had errors in it, and so one needs to separate case of the file paths being hard-coded at export-time from other cases of missing files which may have a cause other than the exporter. @zhiyong80 If you di find how this went wrong as a separate cause, please do raise a fresh issue.

As noted, in this thread, its not just Keil exports, the same issue occurs for export gcc makefile.  @Navis-Raven ? Did the editing the project solution resolve this for you? If So I'd like to close this ticket, and refer people going forward to the more current ticket https://github.com/ARMmbed/mbed-os/issues/8277 , so we can focus effort in one place going forward.

Navis-Raven, 2019-03-04:
@ConradBraam 
Hi, I don't use Keil µV5 (or ARM mbed online compiler) because of these problems I had. I've uninstalled Keil µV5 since then. I've switched to VSCode with platformIO. So unfortunately I can't help you for the tests. If you want to test a solution you just have to repeat the protocol

zhiyong80, 2019-03-04:
@ConradBraam you are right there was indeed something else. I deleted a very old bin file on mbed drive thinking of it was not needed because I was downloading code from Keil anyway. That turned out to be the cause of my problem which was downloaded code didn't run after downloading.
Regarding the issue of this thread, I did and still do see the error, but that seems not to affect building nor built code. So I will leave with it for now.

Kabron287, 2019-05-27:
Project HSP_RPC_GUI for MAX32620. Mbeg creates a huge single   string include record (about 2.5kB) for Keil include paths. Obviously, compiler could not handle it and that results in many "file not found" errors.
If manually recreate all include records one record per include it works.

Kabron287, 2019-05-27:
Mbed sets incorrect include paths:
`;/HSP;/HSP/Devices;/HSP/Devices/BMP280;/HSP/Devices/BMP280/BMP280;/HSP/Devices/HspLed;/HSP/Devices/HspLed/HspLed;/HSP/Devices/LIS2DH;/HSP/Devices/LIS2DH/LIS2DH;/HSP/Devices/MAX14720;/HSP/Devices/MAX14720/MAX14720;/HSP/Devices/MAX30001;/HSP/Devices/MAX30001/MAX30001;/HSP/Devices/MAX30101;/HSP/Devices/MAX30101/MAX30101;/HSP/Devices/MAX30205;/HSP/Devices/MAX30205/MAX30205;/HSP/Devices/PushButton;/HSP/Devices/S25FS256;/HSP/Hsp_BLE;/HSP/Hsp_BLE/BluetoothLE;/HSP/Interfaces;/HSP/LoggingService;/HSP/QuadSPI;/HSP/RpcServer;/HSP/System;/HSP/Test;/USBDevice;/USBDevice/USBAudio;/USBDevice/USBDevice;/USBDevice/USBHID;/USBDevice/USBMIDI;/USBDevice/USBMSD;/USBDevice/USBSerial;/usr/src/mbed-sdk;mbed-os;mbed-os/cmsis;mbed-os/drivers;mbed-os/events;mbed-os/events/equeue;mbed-os/features;mbed-os/features/FEATURE_BLE;mbed-os/features/FEATURE_BLE/ble;mbed-os/features/FEATURE_BLE/ble/services;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/hci;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/hci/maxwsn;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/stack;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/stack/cfg;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/stack/include;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/util;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/wsf;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/wsf/generic;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/wsf/include;mbed-os/features/frameworks;mbed-os/features/frameworks/greentea-client;mbed-os/features/frameworks/greentea-client/greentea-client;mbed-os/features/frameworks/unity;mbed-os/features/frameworks/unity/unity;mbed-os/features/frameworks/utest;mbed-os/features/frameworks/utest/utest;mbed-os/features/mbedtls;mbed-os/features/mbedtls/inc;mbed-os/features/mbedtls/inc/mbedtls;mbed-os/features/mbedtls/platform;mbed-os/features/mbedtls/platform/inc;mbed-os/features/netsocket;mbed-os/hal;mbed-os/hal/hal;mbed-os/hal/storage_abstraction;mbed-os/platform;mbed-os/rtos;mbed-os/rtos/rtx/TARGET_CORTEX_M;mbed-os/targets/TARGET_Maxim;mbed-os/targets/TARGET_Maxim/TARGET_MAX32620;mbed-os/targets/TARGET_Maxim/TARGET_MAX32620/TARGET_MAX32620HSP;mbed-os/targets/TARGET_Maxim/TARGET_MAX32620/device
`
**Correct paths is:**
`.\HSP\Interfaces;.\HSP\QuadSPI;.\HSP\LoggingService;.\HSP\Devices\MAX30001\MAX30001;.\HSP\Devices\LIS2DH\LIS2DH;.\HSP\Devices\HspLed\HspLed;.\HSP\RpcServer;.\HSP\System;.\HSP\Devices\MAX30205\MAX30205;.\HSP\Devices\S25FS256;.\USBDevice\USBDevice;.\HSP\Hsp_BLE\BluetoothLE;.\HSP\Devices\PushButton;.\HSP\Hsp_BLE;.\HSP\Devices\LIS2DH;.\HSP\Devices\MAX30001;.\HSP\Devices\BMP280;.\HSP\Devices\HspLed;.\HSP\Test;;/HSP;/HSP/Devices;/HSP/Devices/BMP280;/HSP/Devices/BMP280/BMP280;/HSP/Devices/HspLed;/HSP/Devices/HspLed/HspLed;/HSP/Devices/LIS2DH;/HSP/Devices/LIS2DH/LIS2DH;/HSP/Devices/MAX14720;/HSP/Devices/MAX14720/MAX14720;/HSP/Devices/MAX30001;/HSP/Devices/MAX30001/MAX30001;/HSP/Devices/MAX30101;/HSP/Devices/MAX30101/MAX30101;/HSP/Devices/MAX30205;/HSP/Devices/MAX30205/MAX30205;/HSP/Devices/PushButton;/HSP/Devices/S25FS256;/HSP/Hsp_BLE;/HSP/Hsp_BLE/BluetoothLE;/HSP/Interfaces;/HSP/LoggingService;/HSP/QuadSPI;/HSP/RpcServer;/HSP/System;/HSP/Test;/USBDevice;/USBDevice/USBAudio;/USBDevice/USBDevice;/USBDevice/USBHID;/USBDevice/USBMIDI;/USBDevice/USBMSD;/USBDevice/USBSerial;/usr/src/mbed-sdk;mbed-os;mbed-os/cmsis;mbed-os/drivers;mbed-os/events;mbed-os/events/equeue;mbed-os/features;mbed-os/features/FEATURE_BLE;mbed-os/features/FEATURE_BLE/ble;mbed-os/features/FEATURE_BLE/ble/services;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/hci;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/hci/maxwsn;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/stack;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/stack/cfg;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/stack/include;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/util;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/wsf;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/wsf/generic;mbed-os/features/FEATURE_BLE/targets/TARGET_Maxim/exactLE/wsf/include;mbed-os/features/frameworks;mbed-os/features/frameworks/greentea-client;mbed-os/features/frameworks/greentea-client/greentea-client;mbed-os/features/frameworks/unity;mbed-os/features/frameworks/unity/unity;mbed-os/features/frameworks/utest;mbed-os/features/frameworks/utest/utest;mbed-os/features/mbedtls;mbed-os/features/mbedtls/inc;mbed-os/features/mbedtls/inc/mbedtls;mbed-os/features/mbedtls/platform;mbed-os/features/mbedtls/platform/inc;mbed-os/features/netsocket;mbed-os/hal;mbed-os/hal/hal;mbed-os/hal/storage_abstraction;mbed-os/platform;mbed-os/rtos;mbed-os/rtos/rtx/TARGET_CORTEX_M;mbed-os/targets/TARGET_Maxim;mbed-os/targets/TARGET_Maxim/TARGET_MAX32620;mbed-os/targets/TARGET_Maxim/TARGET_MAX32620/TARGET_MAX32620HSP;mbed-os/targets/TARGET_Maxim/TARGET_MAX32620/device;.\HSP\Devices\BMP280\BMP280;.\USBDevice\USBSerial;.\HSP\Devices\MAX14720\MAX14720;.\HSP\Devices\MAX30101\MAX30101;.\HSP\Devices\MAX30101;.\HSP\Devices\MAX30205;.\HSP\Devices\HspLed;.\HSP\Devices\MAX14720
`

0xc0170, 2019-05-27:
@Kabron287 Thanks for sharing. What is difference in the above? I see only  `.\` added  ? I recall we used to have it but was refactored (shorten paths, not needed).

> If manually recreate all include records one record per include it works.

What is "one record" in uvision project file?

Kabron287, 2019-05-27:
**@Martin**
I do not know what the difference. I am quite satisfied that it works.
And with mbed notation, even attempt to edit this record causes Keil crash.
"One record" means the following
[image]

Kabron287, 2019-05-27:
So, the fast solution is:
1. Copy the whole include path string from the **C/C++** tab to clipboard(RightMouseClick Select All, Copy)
2. Paste the sting into editor
3. in Editor Ctrl+H search for **;/** replace to **;.\** Copy string back to clipboard
4. Paste clipboard content back in the **C/C++** tab
5. Enjoy

0xc0170, 2019-05-27:
@Kabron287 Can you report this to Mbed OS repository, this is useful. I haven't seen this error yet, so would like to step-by-step how to reproduce and fix as above. Tools team will review

Kabron287, 2019-05-28:
I did it.

linlingao, 2019-05-30:
Looks like @Kabron287 opened this one:
https://github.com/ARMmbed/mbed-os/issues/10682

I'll be closing this issue as a duplicate. We'll track it under mbed-os issue.

closed 2019-05-31 by 0xc0170