ARMmbed/mbed-os-example-blinky #134: Blinky example export encounter several issues on VSCode

Navis-Raven, 2018-08-15:
Before, all of that I've installed 
VSCode, 
gcc-arm-none-eabi-7-2018-q2-update-win32-sha2,
pyOCD (with pip)
C/C++ plugin (by Microsoft)  for VSCode

And as described here https://os.mbed.com/docs/v5.9/tutorials/visual-studio-code.html

I followed these instructions: 
Exporting a project/Online Compiler
Configuring the debugger

But at this step "Open the .vscode/launch.json file"
(1) There is the first problem:
There is no .vscode folder launch.json is outside the folder. 

I pushed F5 key to debug, in order to let the VSCode program create this folder. And I replaced the created lauch.json file, by ours. It solved the problems of compilation. 

I continued following the steps until "Debugging your project"

(2) Here is the second issue, and it is in the documentation : 
[image]

But there is no debug tab by default ! 
[image]

You'have to click on view then Debug to make it appear. 

After this I clicked on the triangle of the debug tab as explained (equivalent of F5), then third issue occurs !

(3) It says that make.exe is unfoundable: 
[image]

I remind that I've already installed GNU GCC tools for arm and added them to the PATH, like this: 
[image]

So here I'm blocked, I don't kown how to go further. 

(4) The fourth problem is that : On the getting started file
[image]

When you click on this : https://os.mbed.com/docs/v5.6/tools/exporting.html/

You get this : 
[image]

Navis-Raven, 2018-08-15:
So we have about 6 issues here at once !

0xc0170, 2018-09-18:
`make` is not part of GCC ARM installation. You need to have that installed separately.

Other issues seems like a problem in the documentation and vs code exporter itself

@janjongboom Can you review?

janjongboom, 2018-09-19:
@0xc0170  Indeed Make is not part of GCC. If the '.vscode' folder is not present then that's a bug in the exporter, because it needs to have created this. @theotherjimmy

Navis-Raven, 2019-10-31:
@0xc0170  what's new about this bug ?

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

closed 2019-11-01 by ciarmcom

janjongboom, 2019-11-01:
For anyone running into this issue, https://github.com/janjongboom/mbed-vscode-generator generates vscode templates that do not require make.

Navis-Raven, 2019-11-01:
@janjongboom  personnall I used platformIO 

But this bug must be solved for any newcommer that doesn't know how to manage these problems.

Navis-Raven, 2019-11-01:
@adbridge  the automated bot closed it again...