ARMmbed/mbed-os-example-blinky #137: Export "not supported" even though it is in matrix

Matzone77, 2018-09-26:
Hi

When I try to run command:
    mbed export -i vscode_gcc_arm -m DISCO_F769NI --profile ..\mbed-os\tools\profiles\debug.json
I get:
    "project.py: error: DISCO_F769NI not supported by vscode_gcc_arm"
even though the supported boards matrix says it should be supported. 

/Anders

0xc0170, 2018-09-26:
I've just updated to 5.10.0, and it does not fail, please test or provide a version where it fails

Matzone77, 2018-09-26:
Ok, so how do I know which mbed os version I am using? 

I did remove my existing mbed-os and imported it again, using "mbed import", I would guess I now have the latest version, 5.10.0. 

Still, I get the same fail though...

0xc0170, 2018-09-26:
```
 610e35ddc - (HEAD -> mbed-os-5.10, upstream/mbed-os-5.10) Merge pull request #8163 from ARMmbed/release-candidate (8 days ago) <Anna Bridge>
```

```
mbed export -i vscode_gcc_arm -m DISCO_F769NI
Scan: mbed-os-example-blinky
Keeping existing launch.json
Keeping existing settings.json
Keeping existing tasks.json

mbed export -i vscode_gcc_arm -m DISCO_F769NI --profile debug
Scan: mbed-os-example-blinky
Keeping existing launch.json
Keeping existing settings.json
Keeping existing tasks.json

 mbed export -i vscode_gcc_arm -m DISCO_F769NI --profile mbed-os\tools\profiles\debug.json
Scan: mbed-os-example-blinky
Keeping existing launch.json
Keeping existing settings.json
Keeping existing tasks.json
```

Using mbed-cli 1.8.2 (I also tested 1.7.5)

0xc0170, 2018-09-26:
You can go to mbed-os folder and use git log to check where is your HEAD (which commit it points to) or use mbed-cli to update for you

Matzone77, 2018-09-28:
Ok, so Iget the following information from git log:

```
commit f1587b50d7d1e36e3536a88dd92884029a1f4f10 (HEAD -> master, origin/master, origin/HEAD)
Merge: b1b639948 d32b454c5
Author: Nir Sonnenschein <nir.sonnenschein@arm.com>
Date:   Fri Sep 28 13:23:55 2018 +0300

    Merge pull request #8262 from juhoeskeli/mbrblockdevice_align

    Align MBRBlockDevice writes to underlying BlockDevice write size
```
So, I guess by looking at when the latest activity occured I can conclude I have the latest version, 5.10?

But why doesn't the following commands show which version I am currently working on:
```
mbed ls
mbed-os (#f1587b50d7d1)
```
or 
```
mbed releases
mbed-os (#f1587b50d7d1)
  * mbed-os-5.1.0
  * mbed-os-5.1.1
  * mbed-os-5.1.2
  * mbed-os-5.1.3
  * mbed-os-5.1.4
  * mbed-os-5.1.5
  * mbed-os-5.10.0
  * mbed-os-5.2.0
  * mbed-os-5.2.1
  * mbed-os-5.2.2
...
```

cmonr, 2018-09-28:
> Still, I get the same fail though...

Can you post the output of the failing command, and the command you're attempting to run?
Could you also share what pip list outputs?

Matzone77, 2018-10-01:
Here it is: (Currently I am using Mbed cli, even though the project is located in a Platform IO folder, hence the need for getting debugging in VS code to work)
```
(...)Projects\USB-serial-exempel>mbed export -i vscode_gcc_arm -m DISCO_F769NI --profile ..\mbed-os\tools\profiles\debug.json -v
[mbed] Working path "C:\Users\u2672860\Documents\PlatformIO\Projects\USB-serial-exempel" (program)
[mbed] Exec "c:\users\u2672860\.platformio\penv\scripts\python.exe -u C:\Users\u2672860\Documents\PlatformIO\Projects\mbed-os\tools\project.py -i vscode_gcc_arm -m DISCO_F769NI --source . --profile ..\mbed-os\tools\profiles\debug.json" in "C:\Users\u2672860\Documents\PlatformIO\Projects\USB-serial-exempel"
Scan: USB-serial-exempel
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S [{matrix,ides}]] [--update-packs] [-E]
                  [--build BUILD_DIR] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--app-config APP_CONFIG] [-z]
                  [--ignore IGNORE]
project.py: error: DISCO_F769NI not supported by vscode_gcc_arm
[mbed] ERROR: "c:\users\u2672860\.platformio\penv\scripts\python.exe" returned error.
       Code: 2
       Path: "C:\Users\u2672860\Documents\PlatformIO\Projects\USB-serial-exempel"
       Command: "c:\users\u2672860\.platformio\penv\scripts\python.exe -u C:\Users\u2672860\Documents\PlatformIO\Projects\mbed-os\tools\project.py -i vscode_gcc_arm -m DISCO_F769NI --source . --profile ..\mbed-os\tools\profiles\debug.json"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

cmonr, 2018-10-09:
Interesting. Will try and look at this tomorrow.

Inside of Arch Linux, all of the following commands worked fine for me:
```
git clone https://github.com/ARMmbed/mbed-os
cd mbed-os
mbed deploy

# Sanity check
mbed compile -t GCC_ARM -m DISCO_F769NI -v

# Actual tests
mbed export -i gcc_arm -m DISCO_F769NI -v
mbed export -i vscode_gcc_arm -m DISCO_F769NI -v
mbed export -i vscode_gcc_arm -m DISCO_F769NI --profile mbed-os/tools/profiles/debug.json -v
```

Some thoughts:
 - Why did you have to climb a directory to reference the profile? 
 - If you're using an mbed-os directory  that isn't in the project root, have you tried symlinking the mbed-os source such that it's effectively in the project root?
 - Do you still have issues if you don't include the --profile parameter?

Matzone77, 2018-10-11:
Thank you!

I finally got it working. Yes, I am using a global mbed-os folder. What solved it for me was simply to add "--source ..\mbed-os" to the export command. (Both with and without the --profile parameter works.)

I guess what confused me was the error "not supported by vscode_gcc_arm", to me it was not intuitive that the command couldn't find the correct path.

Thanks again for your dedicated help :)
/Anders

closed 2018-10-11 by cmonr