ARMmbed/mbed-os-example-blinky #146: Compile error in online compiler, report_status.h

shimniok, 2018-11-30:
Upon creating a new program within the online compiler (My target is _mbed LPC1768_), selecting the mbed-os-example-blinky, and making no changes to the example, compilation fails with the following:

`Error: Identifier "MBED_MAX_MEM_REGIONS" is undefined in "stats_report.h", Line: 45, Col: 30
`
In stats_report.h, if I comment out the following for loops that use this define, the program compiles fine:
```
        for (int i = 0; i < MBED_MAX_MEM_REGIONS; i++) {
            if (sys_stats.ram_size[i] != 0) {
                printf("RAM%d: Start 0x%lx Size: 0x%lx \r\n", i, sys_stats.ram_start[i], sys_stats.ram_size[i]);
            }
        }
        for (int i = 0; i < MBED_MAX_MEM_REGIONS; i++) {
            if (sys_stats.rom_size[i] != 0) {
                printf("ROM%d: Start 0x%lx Size: 0x%lx \r\n", i, sys_stats.rom_start[i], sys_stats.rom_size[i]);
            }
        }
```
It looks like MBED_MAX_MEM_REGIONS was added to mbed_stats.h in commit [ad736e989466da375e20cf776409225f25045854](https://github.com/ARMmbed/mbed-os/commit/ad736e989466da375e20cf776409225f25045854#diff-b02d20d66e6ec90bb5e94f9d49f4dd63) and this appears to be in the 5.11-rc1 branch.

So my guess here is that mbed-os-blinky merged in code intended to work with 5.11 but which fails with 5.10.whatever.

cmonr, 2018-11-30:
@shimniok Thanks for the heads up.

We found out about this issue earlier today, and it led to a process review of how we manage examples during the release period.

We'll be fixing the problem tonight.

cmonr, 2018-11-30:
@shimniok Could you try importing the example again in the online compiler? The issue should now be resolved.

closed 2018-12-01 by cmonr