ARMmbed/mbed-os-example-blinky #151: not able to run test_cmdline

mukhia88, 2018-12-21:
**Am I missing something below? The "-vv" very_verbose mode is also not printing all the infos.**

C:\Users\[User]\workspace\mbedOS\mbed-os-example-blinky\mbed-os>mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED -vv
[mbed-4916] Working path "C:\Users\BTEM7\workspace\mbedOS\mbed-os-example-blinky\mbed-os" (library)
[mbed-4916] Program path "C:\Users\BTEM7\workspace\mbedOS\mbed-os-example-blinky"
[mbed-4916] Exec "c:\python27\python.exe -m pip list -l" in "C:\Users\BTEM7\workspace\mbedOS\mbed-os-example-blinky\mbed-os"
Package                       Version
----------------------------- -----------
appdirs                       1.4.3
argcomplete                   1.9.4
asn1ate                       0.6.0
asn1crypto                    0.24.0
beautifulsoup4                4.6.3
certifi                       2018.11.29
cffi                          1.11.5
chardet                       3.0.4
colorama                      0.3.9
coloredlogs                   10.0
cryptography                  2.1.4
Deprecated                    1.2.4
ecdsa                         0.13
enum34                        1.1.6
fasteners                     0.14.1
funcsigs                      1.0.2
functools32                   3.2.3.post2
future                        0.16.0
futures                       3.2.0
fuzzywuzzy                    0.17.0
hgapi                         1.7.4
humanfriendly                 4.17
icetea                        1.1.0
idna                          2.7
intelhex                      2.2.1
intervaltree                  2.1.0
ipaddress                     1.0.22
Jinja2                        2.10
jsonmerge                     1.5.2
jsonpointer                   1.14
jsonschema                    2.6.0
junit-xml                     1.8
lockfile                      0.12.2
Logbook                       1.4.1
lxml                          4.2.5
manifest-tool                 1.4.6
MarkupSafe                    1.1.0
mbed-cli                      1.8.3
mbed-cloud-sdk                2.0.1
mbed-flasher                  0.9.2
mbed-greentea                 0.2.24
mbed-host-tests               1.1.2
mbed-ls                       1.6.2
mbed-test-wrapper             1.0.0
mock                          2.0.0
monotonic                     1.5
ntfsutils                     0.1.4
pathlib                       1.0.1
pbr                           5.1.1
pip                           18.1
prettytable                   0.7.2
project-generator             0.8.17
project-generator-definitions 0.2.38
protobuf                      3.5.2.post1
psutil                        5.4.8
py                            1.7.0
pyasn1                        0.2.3
pycparser                     2.19
pyelftools                    0.23
PyGithub                      1.43.4
PyJWT                         1.7.1
pyocd                         0.14.3
[... 92 lines cut ...]
Requirement already satisfied: funcsigs>=1; python_version < "3.3" in c:\python27\lib\site-packages (from mock->mbed-greentea<=1.5,>=0.2.24->-r requirements.txt (line 13)) (1.0.2)
Requirement already satisfied: pbr>=0.11 in c:\python27\lib\site-packages (from mock->mbed-greentea<=1.5,>=0.2.24->-r requirements.txt (line 13)) (5.1.1)
Requirement already satisfied: lxml in c:\python27\lib\site-packages (from pyshark-legacy->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (4.2.5)
Requirement already satisfied: py in c:\python27\lib\site-packages (from pyshark-legacy->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (1.7.0)
Requirement already satisfied: trollius==1.0.4 in c:\python27\lib\site-packages (from pyshark-legacy->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (1.0.4)
Requirement already satisfied: logbook in c:\python27\lib\site-packages (from pyshark-legacy->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (1.4.1)
Requirement already satisfied: pyOCD!=0.13.0 in c:\python27\lib\site-packages (from mbed-flasher==0.9.*->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (0.14.3)
Requirement already satisfied: pycparser in c:\python27\lib\site-packages (from cffi>=1.7; platform_python_implementation != "PyPy"->cryptography>=2.0.0->manifest-tool==1.4.6->-r requirements.txt (line 20)) (2.19)
Requirement already satisfied: futures in c:\python27\lib\site-packages (from trollius==1.0.4->pyshark-legacy->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (3.2.0)
Requirement already satisfied: websocket-client in c:\python27\lib\site-packages (from pyOCD!=0.13.0->mbed-flasher==0.9.*->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (0.54.0)
Requirement already satisfied: pyusb>=1.0.0b2 in c:\python27\lib\site-packages (from pyOCD!=0.13.0->mbed-flasher==0.9.*->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (1.0.2)
Requirement already satisfied: pywinusb>=0.4.0 in c:\python27\lib\site-packages (from pyOCD!=0.13.0->mbed-flasher==0.9.*->icetea<1.1,>=1.0.2->-r requirements.txt (line 22)) (0.4.2)
valinor 0.0.15 has requirement pyelftools==0.23, but you'll have pyelftools 0.25 which is incompatible.
Installing collected packages: pyelftools, icetea
  Found existing installation: pyelftools 0.23
    Uninstalling pyelftools-0.23:
      Successfully uninstalled pyelftools-0.23
  Running setup.py install for pyelftools ... done
  Found existing installation: icetea 1.1.0
    Uninstalling icetea-1.1.0:
      Successfully uninstalled icetea-1.1.0
Successfully installed icetea-1.0.2 pyelftools-0.25

Please help me with above issue as I cannot say exactly what is the problem.
Thanks in advance!!

0xc0170, 2018-12-21:
It seems you want to test from the example. This is not currently supported (multiple mains collision). If you fetch only mbed-os release , you should be able to run the test.

There's an issue in mbed-cli repository for this support, I can find the tracking issue if that would help?

mukhia88, 2018-12-24:
Hi , Thank you for the reply. I'm not able to view the below support ticket.

https://jira.arm.com/browse/MBOCUSTRIA-359

"If you fetch only mbed-os release , you should be able to run the test." --> Can you please explain me how to do this... Sorry I'm not still clear!!

cmonr, 2018-12-27:
@SeppoTakalo @OPpuolitaival Could y'all help here? This appears to be an issue with running icetea.

@mukhia88 Correct. The URL is internal only for tracking purposes.
Could you humor me and copy the mbed-os directory so that it's standalone instead of being present in a project, and rerun the command again?

mukhia88, 2018-12-28:
@cmonr I don't know why but now I'm able to run the command after copying the ".mbed" file from the root directory to "mbed-os" folder of "mbed-os-example-blinky". Is there any reason why it worked and if there is any other solution that would still help. Below is the mbed-os directory: 

.astyleignore
.astylerc
.coveragerc
.git
.gitattributes
.github
.gitignore
.mbed
.pylintrc
.travis.yml
astyle-branch.out
BUILD
cmsis
components
CONTRIBUTING.md
docs
doxyfile_options
DOXYGEN_FRONTPAGE.md
doxygen_options.json
drivers
events
features
hal
Jenkinsfile
LICENSE
log
logo.png
mbed.h
platform
README.md
requirements.txt
rtos
targets
TESTS
TEST_APPS
tools
UNITTESTS

Thanks!!

OPpuolitaival, 2018-12-28:
`mbed test --icetea` is searching TEST_APPS folder from current path. If that folder founds then it can compile and run the test. Test_cmdline is very simple tests and do not test application. Test application code https://github.com/ARMmbed/mbed-os/blob/master/TEST_APPS/device/exampleapp/main.cpp and test script https://github.com/ARMmbed/mbed-os/blob/master/TEST_APPS/testcases/example/test_cmdline.py

cmonr, 2018-12-28:
@mukhia88 This problem sounds familiar.

A related issue was logged in mbed-os back when `icetea` was introduced: https://github.com/ARMmbed/mbed-os/issues/8064

If memory serves, `mbed-cli`'s behavior is to move to the project's root before running commands. I think copying the `.mbed` file _tricks_ mbed-cli to thinking that the `mbed-os` directory is the project root instead of mbed-os-example-blinky.

SeppoTakalo, 2018-12-31:
Is there now difference on behaviour of `mbed test --greentea` and `mbed test --icetea` and if so, this has to be fixed.

If Icetea tests are limited to be running in `mbed-os` root folder, then the tool should take care of switching the directory to that, before trying the tests.

mukhia88, 2019-01-03:
@OPpuolitaival is there any BLE example applications which runs on handler mode? I see most of there are in thread mode and I'm not able to do much from icetea test cases available. I wanted to write some icetea test cases and run on BLE example sample applications but not able to do so. I took the applications from below location:
https://github.com/ARMmbed/mbed-os-example-ble

@cmonr i saw the ticket and the issue looks similar. I don't know why it did not print error message like it was not in a root folder or something. Then I would have come to know. Anyways I'm able to run icetea test cases for now. 

@SeppoTakalo I have not tried with --greentea yet as the requirement was only icetea.

OPpuolitaival, 2019-01-04:
@donatieng can you answer to the ble question?

pan-, 2019-01-08:
> is there any BLE example applications which runs on handler mode? 

@mukhia88 Could you clarify what your issue is ? 
AFAIK all the examples _runs_ in handler mode. When an event in the thread worlds needs attention, it is passed to the event queue that process them in handler mode.

paul-szczepanek-arm, 2019-01-08:
@mukhia88 about the BLE examples: all supported examples are in that repo so the answer is no

mukhia88, 2019-01-10:
@pan- I want to write an icetea testcase where I will be able to issue a command like connect or disconnect to the example application through the script. Right now like you said all the applications are running in thread mode where it connects and disconnects automatically after the .bin is flashed to the board. Is there a way where I can issue a connect command to the application and receive the connect status event and tell that my testcase is passed? Or do you have any icetea testcases where I can exercise the above in thread mode?

linlingao, 2019-05-30:
It seems the original issue is about "mbed test". It then morphed into a question about BLE application.

0xc0170, 2019-06-04:
I am closing this, if there are questions about BLE/testing, it should be done in support forums or Mbed OS repository.

closed 2019-06-04 by 0xc0170