ARMmbed/mbed-os-example-blinky #153: Default baudrate - 115200

deepikabhavnani, 2019-01-04:
Why is default baud rate for blinky set as 115200? In mbed-os we have default for serial as 9600

cmonr, 2019-01-04:
@kegilbert

kegilbert, 2019-01-04:
I set it higher due to the high amount of serial output causing a noticeable delay lag due to the simple reporting method (printf's after the wait call in main rather than in their own reporting thread). It also provided an introduction to the Mbed OS config system. I'm open to alternatives if you think it adds more confusion than anything else for new users.

deepikabhavnani, 2019-01-04:
> I set it higher due to the high amount of serial output causing a noticeable delay lag due to the simple reporting method

Should we print all stats just once, instead in loop or once for every 100 blink operations?
Just thinking of options, no strong preference

kegilbert, 2019-01-04:
Reducing the print rate would probably be good, it's a bit fast as is.

deepikabhavnani, 2019-01-04:
> Reducing the print rate would probably be good, it's a bit fast as is.

See if https://github.com/ARMmbed/mbed-os-example-blinky/pull/155 is acceptable