ARMmbed/mbed-os-example-blinky #167: [Mbed OS 5.12 OoB] RAM size is too big when compiled with ARM Compiler 6

coisme, 2019-03-20:
This issue is related to Mbed OS 5.12.0 OoB campaign.

When this example program is compiled with ARM, the result binary file size is too large compared to other compiler, i.e. GCC_ARM or IAR.

Here're the result:
ARM Compiler 6 + K64F and bare-metal option
```
| Module           |         .text |   .data |            .bss |
|------------------|---------------|---------|-----------------|
| [lib]\c_w.l      | 10651(+10651) | 12(+12) |       280(+280) |
| [lib]\fz_wm.l    |       34(+34) |   0(+0) |           0(+0) |
| [lib]\m_wm.l     |       48(+48) |   0(+0) |           0(+0) |
| anon$$obj.o      |       32(+32) |   0(+0) | 196752(+196752) |
| main.o           |   1121(+1121) |   0(+0) |         12(+12) |
| mbed-os\cmsis    |   1021(+1021) |   0(+0) |           0(+0) |
| mbed-os\drivers  |     186(+186) |   0(+0) |           0(+0) |
| mbed-os\hal      |   1664(+1664) |   8(+8) |       130(+130) |
| mbed-os\platform |   4530(+4530) | 64(+64) |       257(+257) |
| mbed-os\targets  | 15362(+15362) | 12(+12) |       337(+337) |
| Subtotals        | 34649(+34649) | 96(+96) | 197768(+197768) |
Total Static RAM memory (data + bss): 197864(+197864) bytes
Total Flash memory (text + data): 34745(+34745) bytes
```
ARM Compiler 6 + K64F (no bare-metal option)
```
| Module           |         .text |     .data |          .bss |
|------------------|---------------|-----------|---------------|
| [lib]\c_w.l      |   11367(+716) |    16(+4) |      348(+68) |
| [lib]\fz_wm.l    |        34(+0) |     0(+0) |         0(+0) |
| [lib]\m_wm.l     |        48(+0) |     0(+0) |         0(+0) |
| anon$$obj.o      |        32(+0) |     0(+0) | 189568(-7184) |
| main.o           |      1121(+0) |     0(+0) |        12(+0) |
| mbed-os\cmsis    |      1021(+0) |     0(+0) |         0(+0) |
| mbed-os\drivers  |      74(-112) |     0(+0) |         0(+0) |
| mbed-os\features |     185(+185) |     0(+0) |     184(+184) |
| mbed-os\hal      |      1664(+0) |     8(+0) |       130(+0) |
| mbed-os\platform |   5844(+1314) |    64(+0) |     361(+104) |
| mbed-os\rtos     |   9970(+9970) | 168(+168) |   6626(+6626) |
| mbed-os\targets  |   15738(+376) |    12(+0) |       337(+0) |
| Subtotals        | 47098(+12449) | 268(+172) |  197566(-202) |
Total Static RAM memory (data + bss): 197834(-30) bytes
Total Flash memory (text + data): 47366(+12621) bytes
```

```
>mbed ls
[mbed] Working path "C:\Users\osakoi01\mbed\mbed-os-example-blinky" (program)
mbed-os-example-blinky (#602de4be56f0, tag: mbed-os-5.12.0-rc2)
`- mbed-os (#85cbc290a2b4, tag: mbed-os-5.12.0-rc3)
```

```
>mbed --version
1.9.2
```

cmonr, 2019-03-20:
@coisme Mind providing us with what "too big" means in this context?

What is the right size? Can you provide the output when using the other two compilers to more easily comare the two?

coisme, 2019-03-21:
@cmonr Hi, I added GCC_ARM and IAR results.

adbridge, 2019-03-21:
@coisme You haven't provided the bare metal options for the other compilers ? (only the no bare metal ones)

coisme, 2019-03-22:
@adbridge Sorry, I added those information in the first comment.

bulislaw, 2019-03-22:
`Total Static RAM memory (data + bss): 197864(+197864) bytes`

I guess static RAM usage is a bit excessive. Can someone investigate why this `| anon$$obj.o      |       32(+32) |   0(+0) | 196752(+196752) |` is being pulled in?

http://infocenter.arm.com/help/topic/com.arm.doc.100070_0612_00_en/pge1362066061445.html

@theotherjimmy @SenRamakri could you guys have a look what's going on?

0xc0170, 2019-03-22:
Are we using base platform model? I dont see any flags in the profiles that would confirm.

bulislaw, 2019-03-22:
¯\_(ツ)_/¯ I don't think we do, but I can't find any occurrence of the `anon$$obj.o` in ARMC manuals.

theotherjimmy, 2019-03-22:
@bulislaw I think that is the heap, and `anon$$obj.o` may be the name that memap gives all regions it does not recognise

bulislaw, 2019-03-22:
Ah! Good point that would make sense. I still don't like how it's accounted there, but at least it's not critical issue...

iriark01, 2019-03-25:
Ashok mentioned this requires a docs note. Can anyone supply that?

SenRamakri, 2019-03-25:
@ashok-rao @iriark01 - I have discussed this with @bulislaw and @adbridge we think this need not go into  known issues, as we think its not an issue really. Its according to the current implementation/design. For ARM toolchain we current create a section for heap in scatter files and thats why its ges accounted for in Anonymous objects. See here please - https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct#L128 
Adding this to known issues may cause confusion. So lets not add it there, instead we will review internally if any fix needs to be investigated. Hope this info helps.

0xc0170, 2019-08-14:
I believe this issue can be closed as noted above. If this comes up again, we should rereview this one.

closed 2019-08-14 by 0xc0170