ARMmbed/mbed-os-example-blinky #201: [OoB 5.15] Cannot clone the repository using the given command

coisme, 2019-12-04:
This issue is related to Mbed OS 5.15 OoB campaign.

When I tried importing project with this command given in README
```
$ git clone git@github.com:armmbed/mbed-os-example-blinky
```
I got this error
```
$ git clone git@github.com:ARMmbed/mbed-os-example-blinky
Cloning into 'mbed-os-example-blinky'...
Enter passphrase for key '/Users/osakoi01/.ssh/id_rsa':
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

Do I need any setting before clone the repository?

I can import the same repository with this command given in README
```
$ mbed import mbed-os-example-blinky
[mbed] Working path "/Users/osakoi01/mbed/20191204_MbedOS5.15_OoB" (directory)
[mbed] Importing program "mbed-os-example-blinky" from "https://github.com/ARMmbed/mbed-os-example-blinky" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #cf4f12a123c0
[mbed] Auto-installing missing Python modules (pywin32, wmi)..
```

0xc0170, 2019-12-04:
@coisme One is using ssh another one https. For ssh (the first case), you don't have ssh agent running (I assume).

We should check other examples, github recommends using https (a user can always use ssh, easy to switch).

adbridge, 2019-12-04:
@coisme https://help.github.com/en/github/authenticating-to-github/connecting-to-github-with-ssh this should be of help

coisme, 2019-12-05:
Hi @0xc0170 and @adbridge Thank you for your information. I followed **Generating a new SSH key and adding it to the ssh-agent** in the above link. Then the command ran successfully.
```
$ git clone git@github.com:armmbed/mbed-os-example-blinky
Cloning into 'mbed-os-example-blinky'...
Warning: Permanently added the RSA host key for IP address '13.114.40.48' to the list of known hosts.
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 716 (delta 1), reused 5 (delta 1), pack-reused 706
Receiving objects: 100% (716/716), 292.50 KiB | 860.00 KiB/s, done.
Resolving deltas: 100% (214/214), done.
```
If we keep the first command, it is better to mention this SSH stuff.

adbridge, 2019-12-05:
@coisme Glad that helped. As this is pretty much Git 101 I don't think it's something we should necessarily be adding to our READMEs .....I will close this now then.

closed 2019-12-05 by adbridge