ARMmbed/mbed-os-example-blinky #239: ThisThread::sleep_for(var) issue

JaMinsane, 2020-09-23:
When I imported the source code into mbedOS online compiler and then tried to compile. I got this error: 

**Error: No matching literal operator for call to 'operator""ms' with argument of type 'unsigned long long' or 'const char *', and no matching literal operator template in "main.cpp", Line: 20, Col: 31**

I didn't change anything but It can't compile.

**main.cpp**

/* mbed Microcontroller Library
 * Copyright (c) 2019 ARM Limited
 * SPDX-License-Identifier: Apache-2.0
 */

#include "mbed.h"

// Blinking rate in milliseconds
#define BLINKING_RATE     500ms

int main()
{
    // Initialise the digital pin LED1 as an output
    DigitalOut led(LED1);

    while (true) {
        led = !led;
        ThisThread::sleep_for(BLINKING_RATE);
    }
}

mbzhu1, 2020-10-15:
I got the same error, but it compiled for me when I removed the "ms" from the `BLINKING_RATE`
Changed: ```#define BLINKING_RATE     500ms```  to ```#define BLINKING_RATE     500```

rajkan01, 2020-11-27:
Hi James,
Thanks for reporting the issue.
This issue caused by `mbed import mbed-os-example-blinky` command which gets invoked at the time of importing the source code into `mbedOS online compiler`, import command is importing an old version of `mbed-os-5.15.4` instead of the latest released version. As the Chrono C++ functionality of passing Chrono duration like `time with ms (500ms)` suffix support is available from `mbed-os-6.0.0`.

Already we have an internal ticket https://jira.arm.com/browse/IOTOSM-278 to investigate `mbed import` command issue.

I will update here, once I have some information on the progress of this issue.

Thanks

evedon, 2020-12-16:
I will close this as I believe your question was answered.

closed 2020-12-16 by evedon