ARMmbed/mbed-os-example-blinky #240: Ignoring hidapi: markers 'platform_system != "Linux"' don't match your environment

symeonmattes, 2020-09-27:
Hi,

I'm running the command
`mbed compile -m EP_AGORA -t GCC_ARM --flash`

and I receive:
```
[mbed] Working path "/home/symeon/Mbed Programs/EmbeddedPlannet/mbed-os-example-blinky" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: six, idna, cryptography
       You can install all missing modules by running "pip install -r requirements.txt" in "/home/symeon/Mbed Programs/EmbeddedPlannet/mbed-os-example-blinky/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
Could not find executable for GCC_ARM.
Currently set search path: No path set
[mbed] ERROR: "/usr/bin/python3" returned error.
       Code: 2
       Path: "/home/symeon/Mbed Programs/EmbeddedPlannet/mbed-os-example-blinky"
       Command: "/usr/bin/python3 -u /home/symeon/Mbed Programs/EmbeddedPlannet/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m EP_AGORA --source . --build ./BUILD/EP_AGORA/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```
Then I run 
`sudo pip install -r mbed-os/requirements.txt`

and I receive:

```
Ignoring hidapi: markers 'platform_system != "Linux"' don't match your environment
Ignoring pywin32: markers 'platform_system == "Windows"' don't match your environment
Ignoring wmi: markers 'platform_system == "Windows"' don't match your environment
```
I'm running python3, pip3 on 

```
sudo lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

symeonmattes, 2020-09-28:
I have managed to do the following:

`sudo pip install -U -r requirements.txt`

This installed the missing modules idna, cryptography, but not six. Then from requirments.txt I saw that "six==1.12.0" So I run

`sudo pip install 'six==1.12.0' --ignore-installed`

The ignore-installed was necessary otherwise I couldn't install it. Now what I receive with

`mbed compile -m EP_AGORA -t GCC_ARM --flash`

```
[mbed] Working path "/home/symeon/Mbed Programs/EmbeddedPlannet/mbed-os-example-blinky/mbed-os" (library)
[mbed] Program path "/home/symeon/Mbed Programs/EmbeddedPlannet/mbed-os-example-blinky"
Could not find executable for GCC_ARM.
Currently set search path: No path set
[mbed] ERROR: "/usr/bin/python3" returned error.
       Code: 2
       Path: "/home/symeon/Mbed Programs/EmbeddedPlannet/mbed-os-example-blinky/mbed-os"
       Command: "/usr/bin/python3 -u /home/symeon/Mbed Programs/EmbeddedPlannet/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m EP_AGORA --source .. --build ../BUILD/EP_AGORA/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

symeonmattes, 2020-09-28:
Ok...finally I've managed to make it work:

I downloaded the [gcc compiler](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) and I configured it by running:

mbed config --global GCC_ARM_PATH <path>/gcc-arm-none-eabi-9-2019-q4-major/bin

Please fix also your documentation to include these commands as well because it took a while to understand what's going on

evedon, 2020-10-02:
Hi @symeonmattes,
Glad that you managed to install mbed cli and got the program to work. We do have documentation https://os.mbed.com/docs/mbed-os/v6.3/build-tools/install-and-set-up.html but I agree that the link is not easy to find.

evedon, 2020-10-02:
@iriark01 Could you look into replacing https://os.mbed.com/docs/mbed-os/v6.3/quick-start/build-with-mbed-cli.html with https://os.mbed.com/docs/mbed-os/v6.3/build-tools/install-and-set-up.html as the quick start guide does not provide information to configure compiler location?

iriark01, 2020-10-02:
Do you still need to configure it when you use an installer?

0xc0170, 2021-06-14:
Closing as it has not been updated since the last year.

closed 2021-06-14 by 0xc0170

symeonmattes, 2021-06-14:
Hi,

Sorry for my late reply, as I didn't notice the comment. Just in case somebody else has the same issue, actually it's a conflict of the pip library. I tried different things such as:

`pip install -r mbed-os/requirements.txt`

The idea is to uninstall the packages i.e. 

`
pip uninstall six, idna, cryptography
`
and then use this

`pip install -r mbed-os/requirements.txt`