ARMmbed/mbed-os-example-blinky #252: Error building this example for DISCO_L475VG_IOT01A

pablogs9, 2021-01-20:
I follow these steps:

```bash
mbed import https://github.com/ARMmbed/mbed-os-example-blinky
cd mbed-os-example-blinky/
mbed compile -m DISCO_L475VG_IOT01A -t GCC_ARM
```

With version:

```bash
vscode@pgarrido:/workspaces/develop_mbed/mbed-os-example-blinky$ mbed --version
1.10.5
```

And I get:
```
[mbed] Working path "/workspaces/develop_mbed/mbed-os-example-blinky" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: cryptography
       You can install all missing modules by running "pip install -r requirements.txt" in "/workspaces/develop_mbed/mbed-os-example-blinky/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
Building project mbed-os-example-blinky (DISCO_L475VG_IOT01A, GCC_ARM)
Scan: mbed-os-example-blinky
Compile [  0.1%]: irq_cm4f.S
Compile [  0.2%]: RTX_Config.c
Compile [  0.3%]: rtx_lib.c
Compile [  0.4%]: rtx_evr.c
Compile [  0.5%]: cmsis_os1.c
Compile [  0.6%]: rtx_delay.c
Compile [  0.6%]: rtx_memory.c
Compile [  0.7%]: rtx_system.c
Compile [  0.8%]: mbed_tz_context.c
Compile [  0.9%]: rtx_semaphore.c
Compile [  1.0%]: rtx_evflags.c
Compile [  1.1%]: rtx_mutex.c
Compile [  1.2%]: rtx_timer.c
Compile [  1.3%]: rtx_kernel.c
[Warning] rtx_kernel.c@93,3: 'memset' offset [17, 164] from the object at 'osRtxInfo' is out of the bounds of referenced subobject 'kernel' with type 'struct <anonymous>' at offset 8 [-Warray-bounds]
Compile [  1.4%]: rtx_mempool.c
Compile [  1.5%]: rtx_msgqueue.c
Compile [  1.6%]: os_tick_ptim.c
Compile [  1.7%]: os_systick.c
Compile [  1.7%]: mbed_boot_gcc_arm.c
Compile [  1.8%]: hci_cmd_bis.c
Compile [  1.9%]: hci_core.c
Compile [  2.0%]: hci_cmd_cis.c
Compile [  2.1%]: hci_cmd_ae.c
Compile [  2.2%]: rtx_thread.c
Compile [  2.3%]: mbed_rtos_rtx.c
Compile [  2.4%]: hci_cmd_cte.c
Compile [  2.5%]: mbed_boot.c
Compile [  2.6%]: hci_cmd_phy.c
Compile [  2.7%]: hci_cmd.c
Compile [  2.8%]: mbed_rtx_handlers.c
Compile [  2.8%]: hci_vs_ae.c
Compile [  2.9%]: hci_cmd_iso.c
Compile [  3.0%]: hci_cmd_past.c
Compile [  3.1%]: sec_ecc_debug.c
Compile [  3.2%]: hci_core_ps.c
Compile [  3.3%]: sec_aes.c
Compile [  3.4%]: sec_aes_rev.c
Compile [  3.5%]: att_uuid.c
Compile [  3.6%]: sec_ccm_hci.c
Compile [  3.7%]: sec_ecc_hci.c
Compile [  3.8%]: sec_cmac_hci.c
Compile [  3.9%]: sec_main.c
Compile [  3.9%]: att_main.c
Compile [  4.0%]: att_eatt.c
[... 738 lines cut ...]
                 from ./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_key_storage.c:22:
./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_key_storage.c: In function 'ws_pae_key_storage_array_time_check_and_update_all':
./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_key_storage.c:972:46: error: expected ')' before 'PRIi64'
  972 |         tr_error("KeyS array time all err: %"PRIi64", ref: %"PRIi64", diff: %"PRIi32, ws_pae_current_time_get(), reference_time, time_difference);
      |                                              ^~~~~~
./mbed-os/platform/mbed-trace/include/mbed-trace/mbed_trace.h:143:79: note: in definition of macro 'tr_error'
  143 | #define tr_error(...)           mbed_tracef(TRACE_LEVEL_ERROR,   TRACE_GROUP, __VA_ARGS__)   //!< Print Error Message
      |                                                                               ^~~~~~~~~~~
./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_key_storage.c:972:18: warning: spurious trailing '%' in format [-Wformat=]
  972 |         tr_error("KeyS array time all err: %"PRIi64", ref: %"PRIi64", diff: %"PRIi32, ws_pae_current_time_get(), reference_time, time_difference);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
./mbed-os/platform/mbed-trace/include/mbed-trace/mbed_trace.h:143:79: note: in definition of macro 'tr_error'
  143 | #define tr_error(...)           mbed_tracef(TRACE_LEVEL_ERROR,   TRACE_GROUP, __VA_ARGS__)   //!< Print Error Message
      |                                                                               ^~~~~~~~~~~
./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_key_storage.c:972:44: note: format string is defined here
  972 |         tr_error("KeyS array time all err: %"PRIi64", ref: %"PRIi64", diff: %"PRIi32, ws_pae_current_time_get(), reference_time, time_difference);
      |                                            ^

[mbed] ERROR: "/usr/bin/python3" returned error.
       Code: 1
       Path: "/workspaces/develop_mbed/mbed-os-example-blinky"
       Command: "/usr/bin/python3 -u /workspaces/develop_mbed/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m DISCO_L475VG_IOT01A --source . --build ./BUILD/DISCO_L475VG_IOT01A/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

evedon, 2021-01-26:
Hi, I am not able to reproduce the issue. I am also using Mbed CLI `1.10.5`
Which versions of the example and mbed-os are you using? You can find out by running `mbed ls`.
You should see:
```
mbed-os-example-blinky (#19c434027ca5, tag: mbed-os-6.6.0)
`- mbed-os (#d6784c3ee6ad, tags: mbed-os-6.6.0, mbed-os-6.6.0-rc1)
```

pablogs9, 2021-01-26:
Solved, I have re-ran in a clean environment and it works. Sorry and thanks.

closed 2021-01-26 by pablogs9

tartavull, 2021-02-16:
I'm facing the exact same compilation problem. I deleted the BUILD folder but the problem persists.
Have you done any additional cleanup @pablogs9 ?

ccu1tn, 2021-04-23:
Hello @pablogs9 ; 
 I am importing the embed from this link: mbed import https://github.com/ARMmbed/mbed-os-example-blinky
But it appears this error. 
[image]

Do you know how to fix it?
Please help me! Many thanks

0xc0170, 2021-04-26:
@ccu1tn This is unrelated to this issue. Create a new issue with more detail information (run -v to get an error, where it comes from).

multiplemonomials, 2021-11-01:
I think this issue might be a bug with the Ubuntu package manager version of arm-none-eabi-gcc: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=953844